sccn/eeglab #257: EMDLAB - missing functions

sdaligau, 2021-02-21:
Hi Support,

I get an error message when I run the process : 
EMDLAB>run EMD>ok
The message says that the function 'runemdtrialbytrial' is not defined.
In the plugin directory I see empty directories EMD SEMD MEMD.
Is this normal?
Thank you in advance
Sebastien

| OS version     | centos7 64|
| Matlab version | R2017b |
| EEGLAB version | v2021.0 |

arnodelorme, 2021-03-08:
We have contacted the author and added a note to the plugin. There is nothing more we can do at this point though.

closed 2021-03-08 by arnodelorme

asdf-tang, 2024-12-02:
>Hi, I have the same problem as you
EMDLAB does not continue to run and I also have empty directories EMD SEMD MEMD.
Do you have a solution for this?
Thank you in advance

some of  ERRORS are listed as follow : 
>>EEGLAB error in function pop_runemd() at line 435: function 'runemdtrialbytrial' corresponding to an input parameter of type 'single' is undefined

>> which runemdtrialbytrial 
'runemdtrialbytrial' not found.

| Windows version | win11 64|
| Matlab version | R2020b |
| EEGLAB version | v2023.1 |

asdf-tang, 2024-12-02:
> Hi Support, 您好，支持人员，
> 
> I get an error message when I run the process :运行该进程时收到一条错误消息： EMDLAB>run EMD>ok MD>MD>OK The message says that the function 'runemdtrialbytrial' is not defined.该消息指出未定义函数 'runemdtrialbytrial'。 In the plugin directory I see empty directories EMD SEMD MEMD.在插件目录中，我看到空目录 EMD SEMD MEMD。 Is this normal? 这正常吗？ Thank you in advance 提前感谢您 Sebastien 塞巴斯蒂安
> 
> | OS version | centos7 64||操作系统版本 |CentOS7 64| | Matlab version | R2017b ||Matlab 版本 |R2017b | | EEGLAB version | v2021.0 ||EEGLAB 版本 |2021.0 版 |

Hi, I have the same problem as you.
EMDLAB does not continue to run and I also have empty directories EMD SEMD MEMD.
Have you already solved this problem?
Thank you in advance 

some of ERRORS are listed as follow :
>>EEGLAB error in function pop_runemd() at line 435: function 'runemdtrialbytrial' corresponding to an input parameter of type 'single' is undefined

>>which runemdtrialbytrial
'runemdtrialbytrial' not found.

| Windows version | win11 64|
| Matlab version | R2020b |
| EEGLAB version | v2023.1 |

sdaligau, 2024-12-02:
Hi, I didn't solve the problem, I used other scripts (ron1818 PhD_code/EMD_EEMD).
Regards