sccn/eeglab #258: STUDY file indexing

arnodelorme, 2021-02-23:
@CPernet

denizdor, 2021-02-23:
Hello Arnaud and Cyril,

Here I am including all the codes for creating the study and then step by step screenshots/explanations of what I am seeing regarding the indexing issue.

Uploading the data and full analysis to Google drive:
1) "test" folder:  folder/filepath for original set files (there are 52 unique files but the study uses 47 of them), precomputed channel measures (per the code below) and individual subject subfolders created during the first level analysis (per the code below).
2) "test.study" : study file
3) "LIMO_test": first level, LIMO/beta group files

MATLABR2020a
EEGlab2021
MacOS BigSur

**1)Create STUDY using .set files (Change the path here for loading/saving) and then save the STUDY. This is the code I get from “eegh” after loading the datasets from the GUI and then creating the STUDY by File>create study>  ‘using all loaded datasets’ from GUI.**
```
EEG = pop_loadset('filename',{'P030_ATB.set' 'P029_KGL.set' 'P028_BM.set' 'P027_MB.set' 'P026_IMS.set' 'P025_GDx.set' 'P023_AJS.set' 'P021_PWH.set' 'P018_ARW.set' 'P017_JRS.set' 'P016_ODH.set' 'P015_PJJ.set' 'P014_JMG.set' 'P013_EGL.set' 'P011_EM.set' 'P010_EMO.set' 'P009_SVB.set' 'P008_SAP.set' 'P007_ARJ.set' 'P006_KBK.set' 'P005_TRH.set' 'P004_DJK.set' 'P003_TJW.set' 'P002_LMK.set' 'HC029_DJK_4bin.set' 'HC028_JaJ_4bin.set' 'HC027_JLJ_4bin.set' 'HC026_RJW_4bin.set' 'HC024_JES_4bin.set' 'HC23_KWI.set' 'HC022_JMI_4bin.set' 'HC021_DJK_4bin.set' 'HC019_HPD_4bin.set' 'HC018_SFO_4bin.set' 'HC017_LLL_4bin.set' 'HC016_TOM_4bin.set' 'HC014_WBM_4bin.set' 'HC012_CKW_4bin.set' 'HC011_HEB_4bin.set' 'HC010HKF.set' 'HC009_LPJ_4bin.set' 'HC008_TLP_4bin.set' 'HC007_ERB_4bin.set' 'HC005_GAJ_4bin.set' 'HC003_CLD_4bin.set' 'HC002_MML_4bin.set' 'HC001_KSM_4bin.set'},'filepath', 'ENTERFILEPATH ');
[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 0,'study',0); 
[STUDY ALLEEG] = std_editset( STUDY, ALLEEG, 'name','test','commands',{{'index' 1 'subject' '60'} {'index' 2 'subject' '59'} {'index' 3 'subject' '58'} {'index' 4 'subject' '57'} {'index' 5 'subject' '56'} {'index' 6 'subject' '55'} {'index' 7 'subject' '53'} {'index' 8 'subject' '51'} {'index' 9 'subject' '48'} {'index' 10 'subject' '47'} {'index' 1 'group' '2'} {'index' 2 'group' '2'} {'index' 3 'group' '2'} {'index' 4 'group' '2'} {'index' 5 'group' '2'} {'index' 6 'group' '2'} {'index' 7 'group' '2'} {'index' 8 'group' '2'} {'index' 9 'group' '2'} {'index' 10 'group' '2'} {'index' 11 'subject' '46'} {'index' 12 'subject' '45'} {'index' 13 'subject' '44'} {'index' 14 'subject' '43'} {'index' 15 'subject' '41'} {'index' 16 'subject' '40'} {'index' 17 'subject' '39'} {'index' 18 'subject' '38'} {'index' 19 'subject' '37'} {'index' 20 'subject' '36'} {'index' 11 'group' '2'} {'index' 12 'group' '2'} {'index' 13 'group' '2'} {'index' 14 'group' '2'} {'index' 15 'group' '2'} {'index' 16 'group' '2'} {'index' 17 'group' '2'} {'index' 18 'group' '2'} {'index' 19 'group' '2'} {'index' 20 'group' '2'} {'index' 21 'subject' '35'} {'index' 22 'subject' '34'} {'index' 23 'subject' '33'} {'index' 24 'subject' '32'} {'index' 25 'subject' '29'} {'index' 26 'subject' '28'} {'index' 27 'subject' '27'} {'index' 28 'subject' '26'} {'index' 29 'subject' '24'} {'index' 30 'subject' '23'} {'index' 21 'group' '2'} {'index' 22 'group' '2'} {'index' 23 'group' '2'} {'index' 24 'group' '2'} {'index' 25 'group' '1'} {'index' 26 'group' '1'} {'index' 27 'group' '1'} {'index' 28 'group' '1'} {'index' 29 'group' '1'} {'index' 30 'group' '1'} {'index' 31 'subject' '22'} {'index' 32 'subject' '21'} {'index' 33 'subject' '19'} {'index' 34 'subject' '18'} {'index' 35 'subject' '17'} {'index' 36 'subject' '16'} {'index' 37 'subject' '14'} {'index' 38 'subject' '12'} {'index' 39 'subject' '11'} {'index' 40 'subject' '10'} {'index' 31 'group' '1'} {'index' 32 'group' '1'} {'index' 34 'group' '1'} {'index' 33 'group' '1'} {'index' 35 'group' '1'} {'index' 36 'group' '1'} {'index' 37 'group' '1'} {'index' 38 'group' '1'} {'index' 39 'group' '1'} {'index' 40 'group' '1'} {'index' 41 'subject' '9'} {'index' 42 'subject' '8'} {'index' 43 'subject' '7'} {'index' 44 'subject' '5'} {'index' 45 'subject' '3'} {'index' 46 'subject' '2'} {'index' 47 'subject' '1'} {'index' 41 'group' '1'} {'index' 42 'group' '1'} {'index' 43 'group' '1'} {'index' 44 'group' '1'} {'index' 45 'group' '1'} {'index' 46 'group' '1'} {'index' 47 'group' '1'}},'updatedat','on','rmclust','off' );
[STUDY ALLEEG] = std_checkset(STUDY, ALLEEG);
CURRENTSTUDY = 1; EEG = ALLEEG; CURRENTSET = [1:length(EEG)];
EEG = eeg_checkset( EEG );
[STUDY EEG] = pop_savestudy( STUDY, EEG, 'filename','test.study','filepath','/ENTERFILEPATH/');
CURRENTSTUDY = 1; EEG = ALLEEG; CURRENTSET = [1:length(EEG)];

```

**2)	GUI--> Load Study from above 'test'-->Make design-->  GUI-->  Edit Study design  2 variables;  var1: group (1-2); var2: type (10-20).**
```
STUDY = std_makedesign(STUDY, ALLEEG, 1, 'name','STUDY.design 1','delfiles','off','defaultdesign','off','variable1','group','values1',{'1','2'},'vartype1','categorical','variable2','type','values2',{'10','20'},'vartype2','categorical','subjselect',{'1','10','11','12','14','16','17','18','19','2','21','22','23','24','26','27','28','29','3','32','33','34','35','36','37','38','39','40','41','43','44','45','46','47','48','5','51','53','55','56','57','58','59','60','7','8','9'});
[STUDY EEG] = pop_savestudy( STUDY, EEG, 'savemode','resave');
```
**3)	Precompute channel measures**
```
[... cut ...] e STUDY), then this does not lead to an issue since  STUDY.datasetinfo.subjects matches with STUDY.limo.subjects in these cases. I can create my study from scratch according to this, but I think being able to add/remove subjects would be challenging, also numbering my subjects according to my original dataset is really helpful when it comes to comparing EEG data with behavirol data. Again I only tried this on my data, and this might be something I am messing up, but I can’t seem to figure out.**

denizdor, 2021-03-06:
Hello Arnaud and Cyril,

I have updated my eeglab/limo through GitHub (via a the links you shared), and tried this again on a small subset of data.
Here is what happens:

This STUDY.datasetinfo:
You can see that subject '3'(first row) has a trial info 1x87:
[image]

Than when I look at Derivatives--> sub-3-->LIMO.mat--> LIMO.design.X, it is 125 which is the. number of trials from subject '1' from STUDY.datasetinfo:
[image]

[image]

When look at   Derivatives--> sub-2-->LIMO.mat--> LIMO.design.X, it is 199 which is the number of trials from subject '2' from STUDY.datasetinfo,  so this one is correct.
When look at   Derivatives--> sub-1-->LIMO.mat--> LIMO.design.X, it is 87 which is the number of trials from subject '3' from STUDY.datasetinfo,  so this one is also incorrect.

*This time pipeline in and out files match, so this is not the error:
pipeline(1).import.files_in: test/3_limo_file_design1_sess1.set
pipeline(1).import.files_out: test/derivatives/LIMO_subj/sub-3/design1_GLM_Channels_Time_OLS/LIMO.mat

*It looks like it follows the order of STUDY.datasetinfo.index, and maybe the data  from subject '3' is named as 1_limo_file_design1_sess1.set since it is index number is 1? 

Thanks a lot! I can also share this small sample of the data with you if it helps!
Deniz

CPernet, 2021-03-07:
we need to see the screenshot of the STUDY because your file names, subject order, subject names are all different (while in BIODS it's all the same) -- which is where issue arises (because the way things get ordered inside STUDY)

closed 2021-03-28 by CPernet