sccn/eeglab #262: matlab and gui freezes when plotting in STUDY

ardian-m, 2021-03-01:
Hello all,

I am a student working with eeglab and recently, I updated to the newest version of eeglab, but I have encountered a problem. I have set up a STUDY with 2 conditions, each containing 5 datasets, and everytime I try to plot measures, MATLAB and the eeglab GUI freezes, this happens everytime I select the measures I want to plot i.e: ERPs or anything else. 

I, also, have tried doing it through a script not GUI but MATLAB somehow still freezes. Can you please help me? In the previous version, it used to work without any problems, has it anything to do with the new version? 

Best regards,

Ardi

arnodelorme, 2021-03-08:
Dear Ardi,

Would you mind trying the previous version and check that it does not freeze.

Also some considerations:
- Sometimes MATLAB gets into a weird state. Have you tried restarting MATLAB?

If the old EEGLAB version does not freeze and the new one does consistently even after restarting Matlab, please send us:
- Operating system
- Matlab version

Cheers,

Arno

ardian-m, 2021-03-09:
Dear Arno,

I have tried it in the previous version (STUDY and plotting that is), it did
work, not flawlessly but after glitching or restarting it did work in the end. 

Yes, I have tried restarting MATLAB, but it still freezes. I use MATLAB 2020a
version, on a Windows 10 OS computer. 

I can use the previous version, it wouldn't be a big deal switching back, but
still wanted to give my feedback.

Best,
Ardi

>>> Arnaud Delorme <notifications@github.com> 08.03.21 22.28 Uhr >>>

 Dear Ardi,
 Would you mind trying the previous version and check that it does not freeze.
 Also some considerations:
  
* Sometimes MATLAB gets into a weird state. Have you tried restarting MATLAB? 
 If the old EEGLAB version does not freeze and the new one does consistently
even after restarting Matlab, please send us:
  
* Operating system 
* Matlab version 
 Cheers,
 Arno

cll008, 2021-05-14:
Hi @ardian-m, are you still encountering this issue? I tried to replicate with MATLAB 2020a on Win10 with the wiki [dataset](https://ucsd.zoom.us/j/94728152660) but did not encounter any glitches. I'm wondering if the bug you encountered is related to something being stored in the dataset.

I'll close the issue for now but please reopen this issue if you have time to replicate the issue or have more comments. Thanks!

closed 2021-05-14 by cll008