sccn/eeglab #264: STUDY.design(STUDY.currentdesign).cases.value not valid with sessions

CPernet, 2021-03-03:
### Description

Working on using sessions in STUDY - the problem arises when you do not have the equal number of sessions and/or use different session number. 

<br>
#### Steps to Reproduce

1. I took the study from [data issue 258](https://drive.google.com/drive/folders/1lSThPxwKaFLNmNTMP3fN7tXLmZAghZKD)
2. STUDY subjects column now reads eg 60, 60, 57,57, 20,20, etc 21,23, 4,12 (note on purpose file names and STIDY differ and STUDY name are numbers not in the natural order, and the last four subjects do not repeat)
3. STUDY session column now reads e.g. 1,2,1,2,1,2, etc 1, 1, 2, 2 (indeed for subjects '21' and '23' recordings come from session 1 while for subjects '4' and '12' recording come from session 2) 

<br>
#### Expected behavior:
STUDY.design(STUDY.currentdesign).cases.value should have values 60, 57, 20, 21,23, 4,12

<br>
#### Actual behavior:
values update is not consistent - hard to see a pattern in what actually happens

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Win10|
| Matlab version | 2020b |
| EEGLAB version |dev|

arnodelorme, 2021-03-08:
I was able to recreate a STUDY with subjects HC and P, but not sure what to do with it. Will talk tomorrow.

CPernet, 2021-03-28:
we solved that - rewriting std_limo tmp file loop

closed 2021-03-28 by CPernet