sccn/eeglab #265: Cleanline and Blinker path broken with EEGLAB 2021.0

CorentinWicht, 2021-03-09:
Dear Arnaud,

I recently updated my old _EEGLAB v14.1.2b_ to the new _EEGLAB 2021.0_ version (many thanks for this update). 
Nevertheless, I am encountering an issue I haven't seen earlier which seems to be specific to some plugins (e.g. _Cleanline 2.0_ and _BLINKER 1.1.2_). 

While both plugins are up-to-date,  correctly loaded at EEGLAB startup and adequately included in MATLAB environment path, running simple commands such as e.g. `checkBlinkerDefaults(struct(), getBlinkerDefaults(EEG));` for BLINKER, returns an error that make it look like the path is not set correctly:
[image]

I recently opened a similar issue on the Cleanline Github page thinking  at first it was specific to one plugin:
https://github.com/sccn/cleanline/issues/3

What makes me think it is an EEGLAB-related issue is that running the following fixes the bug:
```
addpath(genpath('E:\GitHub\autoERP\Functions\eeglab2021.0\plugins\Cleanline2.00'))
addpath(genpath('E:\GitHub\autoERP\Functions\eeglab2021.0\plugins\blinkerv1.1.2'))
```

Did anyone report this issue already?

Best,

Corentin

arnodelorme, 2021-03-11:
It is interesting. These two plugins have issues with the internal version being different than the exposed one. I fixed it and also put Blinker on the EEGLAB plugin manager. However, I could not find the problem. The paths are added properly for me.

arnodelorme, 2021-03-11:
I have tried on Windows now and cannot see the problem either. I think it might have to do with a strange path configuration. Usually, you should just start EEGLAB and all the paths will be added. Let me know if you get more insight as to what the problem might be. I am closing this for now, but feel free to reopen.

closed 2021-03-11 by arnodelorme

enricovara, 2022-05-12:
Hi!

I have this issue on a Mac with MATLAB R2019b and EEGLAB v2022.0

Please let me know if I can help solve this by providing more information.

Best wishes,
Enrico

arnodelorme, 2022-05-18:
Hi Enrico,
Thank you for reviving that. Can you try again now. I have updated the Blinker plugin.