sccn/eeglab #269: pop_loadset back compatibility

campusge, 2021-03-16:
When I try to load a dataset using the command pop_loadset('full path of the datset')  it is seems not working  on EEGLab 2021.0 (it does not find the fdt file due a problem of file path, while it is still working perfectly with older EEGLab versions). this way of the command is used also in different EEGLab plugins and Matlab toolboxes working with EEGLab. I'm doing something wrong?
If not, may I suggest to replace in pop_loadset

    if ~strcmpi(inputname, 'filename') && ~strcmpi(inputname, 'filepath') && ~strcmpi(inputname, 'eeg') && ~strcmpi(inputname, 'loadmode') && ~strcmpi(inputname, 'check')
        options = { 'filename' inputname, 'filepath',inputpath};
...

with something like

if ~strcmpi(inputname, 'filename') && ~strcmpi(inputname, 'filepath') && ~strcmpi(inputname, 'eeg') && ~strcmpi(inputname, 'loadmode') && ~strcmpi(inputname, 'check')
       
        [fpath,fname,fext] = fileparts(inputname);
        inputpath = fpath;
        inputname = [fname,fext];
        options = { 'filename' inputname, 'filepath',inputpath};

which seems to work?
Thank you!

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [windows10 64 bit |
| Matlab version | [2019b] |
| EEGLAB version | [2021.0] |

arnodelorme, 2021-03-16:
Thank you for reporting that problem.
I am not able to reproduce the problem. Would you mind sharing your .set and .fdt file? You can put them in a ZIP archive and drag and drop them here (if the file is small).

campusge, 2021-03-24:
Some more info: it seems that the problem could be related to the new mode of saving a single file for datasets. when you use the new mode it is ok. when you use the old split between .set and fdt, if you launch pop_loadset from within the same directory containing both .set and .fdt it is still ok. instead if you are e.g. into eeglab directory and you lounch pop_loadset pointing to a different directory, it seems that it cannot find the .fdt file (which is actually regularly present). in eeglab up to 2020.0 instead everithing works in all cases.

arnodelorme, 2021-03-24:
I have tried.

1. Saved a dataset in 2 files (.set and .fdt)
2. Go to a random folder
3. Load the file from the GUI

It worked. Did you do anything different? Let's get to the bottom of this. It is important. Thank you for your assistance.

campusge, 2021-03-25:
yep also for me it is working from the GUI, the problem occurred when I tried the same thing using the command line (or when scripting), when your curent folder is folderA, having the fdt and set in folderB, typing in command line pop_loadset using the full path with extension of the set file, e.g. EEG = pop_loadset(fullfile(folderB,[name_file_set,'.set'])) it gives me the error

arnodelorme, 2021-03-25:
OK, I was able to reproduce the problem.

campusge, 2021-03-25:
good. thank you very much for your patience. The problem does not occur if you use the command pop_loadset([name_file_set,'.set'], 'filepath',folderB ), while the full path of the file was earlier working as a unique argument, now it seems that the path is removed at a certain point (maybe in the checkset?) and so the function is assuming that the fdt is in the working directory (not in the directory with the set file). the lines of code proposed try to "recover" the path.

arnodelorme, 2021-03-25:
OK, I have fixed it in 4faabcc35d172704298a7316db00ac77855a8b23
If you pull the latest changes, it should work.

closed 2021-03-25 by arnodelorme