sccn/eeglab #274: EEGLAB: Error in function pop_runica()

RaphaelCertain, 2021-03-26:
### Description

[Hello,
I am conducting pre-processing on EEG recordings and when trying to "Decompose data by ICA" I get an error message indicating "_EEGLAB error in function pop_runica() at line 412 : 
Output argument "sphere" (and maybe others) not assigned during call to "runica"_."
Did you already had this error before? 

Thank you very much,
Raphael ]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2021-04-23:
Apologies for the late reply. We have not seen this error before. This is likely because the data is empty or contains 0 (at least one channel) (my best guess). Send us your data if this is not the case, and we will have a look.

evastone97, 2022-03-30:
I have the same issue now.. do we finally know what was the problem?

arnodelorme, 2022-03-30:
Could you share your dataset and how to reproduce the problem?

arnodelorme, 2022-04-19:
We need a dataset to reproduce the problem. I am closing this report, but feel free to reopen it.

closed 2022-04-19 by arnodelorme

evastone97, 2022-04-19:
I found the problem so now is working ok :) thank you though!

BFutrell03, 2022-04-19:
Care to share your solution evastone97?

zaeemhadi, 2024-05-04:
EEGLAB v2024.0
I came across this error when I tried to run Robust Extended Infomax for ICA decomposition. 

Upon clicking this option, some default parameters are automatically selected ('extended', 1, 'lrate', 1e-5, 'maxstep', 2000). 
Changing 'maxstep', 2000 to 'maxsteps', 2000 fixed the issue for me. 

as used here: [](https://sccn.ucsd.edu/wiki/Makoto%27s_useful_EEGLAB_code#How_to_obtain_practically_reproducible_ICA_results_.2809.2F26.2F2022_added.29%22%3Ereference%3C/a%3E%20for%20ICA%20conservative%20parameters)

arnodelorme, 2024-05-04:
Thank you for reporting.
@MakotoMiyakoshi would you mind to fix your wiki entry.

zaeemhadi, 2024-05-04:
The wiki entry seems to have correct options **"'extended', 1, 'lrate', 1e-5, 'maxsteps', 2000"**

The default options in the eeglab should have 'maxsteps', 2000 instead of 'maxstep', 2000.

This is line 116 of the code in pop_runica.m
[https://github.com/sccn/eeglab/blob/develop/functions/popfunc/pop_runica.m](url)

Best wishes

MakotoMiyakoshi, 2024-05-07:
Hi Zaeemhadi and Arno,

Thank you for reporting the error. I fixed it according to your suggestion.

Makoto

arnodelorme, 2024-05-07:
tx

zaeemhadi, 2024-05-07:
Dear both,
Thank you for responding.

I just wanted to reiterate, "maxstep" is causing this error.
Changing it to "maxsteps" fixed this error for me.

I noticed that Makoto changed the wiki entry to "maxstep". However this doesn't work for Robust extended infomax ICA.

Could you kindly confirm. I might be missing something. 

Best wishes
Zaeem

MakotoMiyakoshi, 2024-05-08:
Hi Zaeem,

Ok I got it. I put it back to 'maxsteps'.
I thought it was strange because I should have copy-and-pasted a working
code.

Makoto