sccn/eeglab #275: Error to use ICLabel

shutli1127, 2021-03-26:
### Description

[*EEGLAB error in function eeg_autocorr_fftw() at line 27: Undefinded function 'resample' for input arguments of type 'double'*]

<br>
#### Expected behavior:
[*Use ICLAbel in EEGLAB to check each independent component after ICA*]

<br>
#### Actual behavior:
[*Error present, but I have installed Signal Processing Toolbox*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*2020a*] |
| EEGLAB version | [*v2020.0*] |

arnodelorme, 2021-04-23:
This is because you need to have the signal processing toolbox. We have added a warning message.

closed 2021-04-23 by arnodelorme

evastone97, 2022-04-03:
what if we have the signal processing toolbox? what is the problem in that case?