sccn/eeglab #282: Apple chip Macbook uses eeglab, the bug appears when saving

furongdan, 2021-04-19:
Matlab 2020b on windows will not report an error when using eeglab, but an error will be reported when using the Apple chip MacBook Matlab2020b.

The specific operation is import data, select data, channel location, re-reference, extract epochs, filter the data, decompose the data by ICA.

when click ''save current data as'', the window''eeglab error'' will be discovered.

thans for your reading.

[image]

furongdan, 2021-04-19:
oooooo,The problem is solved, just delete that folder.

arnodelorme, 2021-04-23:
Looping @robertoostenveld here. The issue is with the overloaded abs function. You should remove the File-IO path for now and see if it works better.

robertoostenveld, 2021-04-24:
Those uint64 mex files should not be used any more with recent MATLAB versions. They are also not released any more with the FieldTrip `fileio` module, but it appears that you are using a version from 2015.  

Mathworks implemented proper support for the uint64 data type already quite some time ago, so it now has its own `abs` function that works on uint64. Indeed just delete the `@uint64` folder.

arnodelorme, 2021-05-03:
So the solution is to update Fieldtrip.

closed 2021-05-03 by arnodelorme