sccn/eeglab #283: event file 'non existent 'latency'' error

Danbeniamini, 2021-04-25:
Hi, I am working with EEG data recorded with EGI.
The data is in a .bin file, and while I manage to open it with pop_fileio, the events arent imported.
Hence I am trying to open the event file from the menu- but I get the error- 'reference to non-existent field 'latency''

Can you help? How should I solve this?

event file (originally xml file uploaded as docx)

[ev.docx](https://github.com/sccn/eeglab/files/6372209/ev.docx)

arnodelorme, 2021-04-25:
Would you mind sharing the data file. Tx Kay.

Danbeniamini, 2021-04-26:
The data is at
https://www.dropbox.com/sh/k7vy5fcf554hf2h/AAB95LHQyCYUWOu63zaMAScqa/raw_data/visual%20detection/EG3_X_test_vis_sc.dsmp/signal1.bin?dl=0

(hope dropbox is ok, couldn't upload as its too big...)

arnodelorme, 2021-04-28:
I have looked at the file, but it does not seem readable using either the EGIPLUGIN (legacy), FILE-IO, or BIOSIG. I would suggest using the MFF EGI format instead. If you absolutely want to import these specific files, I can point towards contractors you could hire to assist you.

closed 2021-04-28 by arnodelorme