sccn/eeglab #286: EEGlab study - plot channel measures (average potentials)

santonial, 2021-04-27:
Good morning,

I am currently working with ERP data using EEGlab study GUI. I want to use the "plot channel measures" option to plot averaged potentials of selected channels and display statistics. If I select up to five channels, the output displays the statistic correctly as in the image:
[image]

However, if I select more than five channels the statistics are not dislayed at all: 
[image]

| OS version     | [Windows 10 |
| Matlab version | R2020.a |
| EEGLAB version | v2021.0 |

I had the same issue with previous versions of EEGLAB.
As I need to work with clusters of channels for display purposes, it would be helpful for me to understand what is the issue.
Thank you,
Alessia

arnodelorme, 2021-04-28:
I understand. This seems like an easy fix. Let us look into it.

arnodelorme, 2021-04-30:
This has been fixed in the GitHub version 3eb7b4b03
Thank you for reporting the problem

closed 2021-04-30 by arnodelorme