sccn/eeglab #287: 'Load existing dataset' uses old directory

tsgouvea, 2021-04-28:
### Description

`Load existing dataset` worked fine once. In the next usage (days later), after using file dialog to select a `.set` file located in a different directory, it fails for searching for the new file in the old directory.

#### Steps to Reproduce

1. Place .set files (and corresponding data files) in two separate directories
e.g. `/dir1/file1.set` and `/dir2/file2.set`
2. Load `/dir1/file1.set` with `Load existing dataset`
3. Close everything (e.g. reboot - in fact, I'm using a fresh installation of Octave the second time around)
4. Load `/dir2/file2.set`  with `Load existing dataset`.

#### Expected behavior:
Load dataset `/dir2/file2.set`

#### Actual behavior:
The following exception is thrown:

> EEGLAB error in function pop_loadset() at line 139:
> 
> load: unable to find file `/dir1/file2.set`
>

#### Versions

|||
|----| ---- | 
| OS version     | Linux Fedora 33|
| Matlab version |Octave 6.2.0|
| EEGLAB version | a7cdc2edbaa45b6a2eb254daa41232fbcb6c33d6|

arnodelorme, 2021-04-28:
Let us have a look

arnodelorme, 2021-07-13:
@cll008 would you mind to have a look

cll008, 2021-07-15:
Hi @tsgouvea are you still encountering this issue? I'm on a Windows machine and could not replicate the problem on either MATLAB or Octave. In step 4 are there any issues when you select `/dir2/file2.set` from the file explorer UI?

Please set a breakpoint here and check what the `inputname` and `inputpath` are
https://github.com/sccn/eeglab/blob/a7cdc2edbaa45b6a2eb254daa41232fbcb6c33d6/functions/popfunc/pop_loadset.m#L66-L67

Could be an issue with [`uigetfile2.m`](https://github.com/sccn/eeglab/blob/40fabb4891bdc48c6a39087d2be07f8b6e65202d/functions/sigprocfunc/uigetfile2.m)

arnodelorme, 2021-07-16:
I have tried as well and could not reproduce the issue. I am closing the issue for now, but feel free to reopen and send us some files if you think this might be useful.

closed 2021-07-16 by arnodelorme