sccn/eeglab #290: STUDY visualisation is slow

Allexxann, 2021-05-06:
Just switched from pre-2019 to the latest version.
I have to plot ERPs for many different channel combinations and designs in one go. Previously it was trivial, each time it took 1-5 minutes to plot, depending on number of channels and files in the STUDY.
With the switch to single-trial it takes forever. I guess, it's because of the size of .daterp files.
Is there any way to speed up the process?
I'm on windows, daterp's from my study collectively weigh about 20.7 Gb.

arnodelorme, 2021-06-15:
The main way to speed up the process is SSD hard drive (solid state drive). For me, it goes at the same speed as before.
You can also decrease the sampling rate. No real need to process your data at more than 250 Hz.

closed 2021-06-15 by arnodelorme

arnodelorme, 2021-06-22:
Dear Allex, Made some fixes that should improve speed. Let us know.

Allexxann, 2023-06-09:
Finally got to update my version. Yes, now the speed is reasonable.