sccn/eeglab #291: Std_readfile error when "data of a subject is split across sessions"

CorentinWicht, 2021-05-07:
### Description 

This issue was already (wrongly) described in the [mp_clustering Github page](https://github.com/sccn/mp_clustering/issues/1).
In fact, the issue is directly related to the behaviour of `std_readfile` when the **data of a subject is split across sessions** (i.e. `if iscell(fileBaseName)`, line 93).

<br>
#### Steps to Reproduce 

The example here includes 3 files (i.e. 3x within-subject factor levels "conditions" for one single subject). 

1. [*First Step*] 
Download the [example dataset](https://www.dropbox.com/s/7n314vm7mlueax2/DataToTest.zip?dl=0) and extract the files.
2. [*Second Step*] 
Make sure the file are in the current MATLAB path and run the following lines: 
`load('DataToTest.mat')`
`data = std_readfile(fileNameAll, 'designvar', design, 'singletrials','off', 'components', compList);`

<br>
#### Expected behavior: 

The `data` output of the function should be a matrix of size ({513×641×105 double}, i.e. 105 being the length of compList. 
Hence, this matrix should be a concatenation on the 3rd dimension of all 3 files. 

<br>
#### Actual behavior: 

The following error is occuring:
[image]

The error may come from line 100 in `std_readfile.m`: 
`  [measureDataTmp{iFile}, parameters, measureRange1, measureRange2, eventsTmp{iFile}] = std_readfile(fileBaseName{iFile}, varargin{:});
`
The main issue is that fileBaseName{iFile} only includes data for 1 of the 3 files, while in varargin the last argument still includes the index of all three files components (i.e. compList). 

<br>
#### Adjustments:

Adjusting the code as follows fix the bug but the output isn't as expected yet (due to the way arrays are concatenated from lines 103-113):

```
if iscell(fileBaseName)
    % this is when the data of a subject is split accross sessions
    % then we need to call std_readfile several times and combine
    % the results
    % ------------------------------------------------------------
     CompList = varargin{end};
     IdxSeq = [find(CompList==1) length(CompList)];
    if length(fileBaseName) > 1
        for iFile = 1:length(fileBaseName)
            varargin{end} = CompList(IdxSeq(iFile):IdxSeq(iFile+1)-1);
            [measureDataTmp{iFile}, parameters, measureRange1, measureRange2, eventsTgmp{iFile}] = std_readfile(fileBaseName{iFile}, varargin{:});
        end
```

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     |  Windows 10 Entreprise Build 18383 |
| Matlab version | R2019b |
| EEGLAB version | 2021.0 |

CorentinWicht, 2021-05-07:
Also I found out regarding the concatenation of arrays that from lines 108 to 110:

```
if ndims(measureDataTmp{iData}{iCell}) == 2, measureData{iCell} = [ measureData{iCell} measureDataTmp{iData}{iCell} ];
elseif ndims(measureDataTmp{iData}{iCell}) == 2, measureData{iCell}(:, :, end+1:end+size(measureDataTmp{iData}{iCell},3)) = measureDataTmp{iData}{iCell};
end
```

the argument of the **if** and **elseif** functions (i.e. `ndims(measureDataTmp{iData}{iCell}) == 2`) are exactly **the same**.

arnodelorme, 2021-05-11:
I have looked at the script and the data you sent. It seems that compList contains the component indices for the 3 sessions. How did you get that component list? Is that from a parent function to std_readfile.

I think the issue might arise from there.

Arno

compList = 

  Columns 1 through 20

     1     2     3     4     5     6     7     8     9    10    11    12    13    14    15    16    17    18    19    20

  Columns 21 through 40

    21    22    23    24    25    26    27    28    29     1     2     3     4     5     6     7     8     9    10    11

  Columns 41 through 60

    12    13    14    15    16    17    18    19    20    21    22    23    24    25    26    27    28    29    30    31

  Columns 61 through 80

    32    33    34    35    36    37     1     2     3     4     5     6     7     8     9    10    11    12    13    14

  Columns 81 through 100

    15    16    17    18    19    20    21    22    23    24    25    26    27    28    29    30    31    32    33    34

  Columns 101 through 105

    35    36    37    38    39

CorentinWicht, 2021-05-11:
> std_readfile

Dear Arno,

Indeed it comes from the parent function `std_readdata`, especially in line 218:
` else [dataTmp{iSubj}, params, xvals, yvals, eventsTmp{iSubj} ] = std_readfile( fileName, 'designvar', struct(bigstruct.design.variable), opts{:}, 'components', compList);`
The `compList` argument being filled and concatenated in the for loop, lines 183-191. 

Additionnally, I am calling `std_readdata` from within `std_mpreclust`, line 191:
` [STUDY specdata] = std_readdata(STUDY,ALLEEG,'clusters',STUDY.etc.preclust.clustlevel,'datatype','spec');`

> Please share your whole study even if it is large. Will be easier for testing.

Sure, here you will find the STUDY and associated files:
https://www.dropbox.com/sh/2uioiv5aoctf3l7/AADz2pGFYJY8hMdDoQMUbc7Ca?dl=0

Best and many thanks for your support,

Corentin

arnodelorme, 2021-05-11:
Thanks Corentin,

I will have a look.

arnodelorme, 2021-05-12:
I have looked into it. Only EEGLAB 14 allows combining ICA decomposition from multiple sessions with different ICA decomposition for the same subject. We discontinued that option because:
- hard to program
- does not make much sense physiologically to compare clusters of components across subjects in this way

closed 2021-05-12 by arnodelorme