sccn/eeglab #292: EETrak errors

Allexxann, 2021-05-07:
### Description

Sometimes people mistakenly configure recorders for this format with tabs instead of spaces (see attachment), and standard readeetraklocs function fails to read the coordinate files made with such configurations.
I've adjusted the function to be able to process both versions correctly and want to share it.

https://1drv.ms/u/s!Avk0-H-bAv5AgQOZLXASK1PSl8bm?e=QLpIYy
https://1drv.ms/u/s!Avk0-H-bAv5AgQJNnaPwylamqjDf?e=i8ngU4

arnodelorme, 2021-06-15:
Thanks. Added the changed in d9fe26d58

closed 2021-06-15 by arnodelorme

Allexxann, 2024-04-15:
Re-opening the issue. Also, I don't know how to contribute properly.

xensor .elc files can include head shape, and this sometimes breaks the function. Or maybe all the times, I'm not sure.
I dug around, in the process I've found mentions of readasalocs, but apparently it was discontinued at some point.

The offending line is
``labels    = locs(indlabels+1:end,:);``
this reads the entirety of the file til the end. If the labels are the last line - you're lucky. Otherwise - not so much.

``labels    = locs(indlabels+1,:);``
This reads only the labels.

There was also a for-if loop, and I replaced this out of principle.

``  locs  = loadtxt( filename );
        
    % get label names
    % ---------------
	
	indlabels=find(strcmpi(locs(:,1),  'Labels'));
	indpos=find(strcmpi(locs(:,1), 'Positions'));
	
    if isempty(indpos) || isempty(indlabels)
        error('Could not find ''Labels'' or ''Position'' tag in electrode file');
	end
	
    labels    = locs(indlabels+1:end,:);
    
    % get positions
    % -------------
    if strcmp(locs(indpos+1,2),':')
        positions = locs(indpos+1:indlabels-1,3:5);
    else
        positions = locs(indpos+1:indlabels-1,1:3);
    end``