sccn/eeglab #296: EEGLAB Study error when computing spectrum and ERSP

elisatatti86, 2021-05-12:
Good evening,
I have created an EEGLAb Study using the resting state recordings from 3 subjects. I need to analyze the resting continuously hence I created a single epoch ranging from 0 to 270s as it seems that EEGLAB study does not compute ERSP without epoched data. 

When I try to precompute the spectrum I get the following error:

_Transpose on ND array is not defined. Use PERMUTE instead._ (error occurred in function std_spec()_  at line 376)

When I try to precompute the ERSP I get this other error:

EEGLAB error in function std_ersp() at line 334. Index exceeds matrix dimensions.

I would really appreciate if you could help me to understand what the problem is. 

Thank you very much in advance. 

Elisa

Allexxann, 2021-05-16:
Not a dev, but ERSP is better suited for epoched data, because it shows an averaged event related dynamic in spectral power.
Resting state usually is analyzed using just spectra, it's a second line, just after ERP. Though they appear to be broken too...

arnodelorme, 2021-07-06:
Spectrum should work on continuous data. ERSP not. Would you mind sharing your data so we can check the issue with computing spectrum?

arnodelorme, 2021-07-16:
Closing for now. Feel free to reopen.

closed 2021-07-16 by arnodelorme