sccn/eeglab #298: Cannot plot ERP for multiltiple channels with multiple sessions design

Allexxann, 2021-05-16:
### Description
Made a study, one variable is "type", another is "session". Precomputed ERPs.
When I plot one channel at a time both variables work fine, in any combination.
When I try to plot several channels (initially I tried "Average them all", later checked other options, same result) at once "type" is working, "session" shows only the first value and swallows everything else as if it didn't exist. Tried to switch off "type" to leave a single variable, doesn't matter.

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2021a] |
| EEGLAB version | [2021.0] |

Allexxann, 2021-05-16:
Since it was urgent I tried to debug it myself.
Traced it to functions\studyfunc\std_readfile.m, lines 108-110 (newlines added for clarity).

**if** ndims(measureDataTmp{iData}{iCell}) == 2
measureData{iCell} = [ measureData{iCell} measureDataTmp{iData}{iCell} ];
**elseif** ndims(measureDataTmp{iData}{iCell}) == 2
measureData{iCell}(:, :, end+1:end+size(measureDataTmp{iData}{iCell},3)) = measureDataTmp{iData}{iCell};
**end**

(0. Matlab says it's better to use ISMATRIX here instead of ndims, but I don't know a difference. Changed it in my copy to get rid of warning and it didn't break anything)
1. if and elseif are completely identical, so the lower branch can never be executed. That's where variable I followed lost the dimension, corresponding to "session".
2. Command on the lower branch adds zero-padded line that somehow messes the variable during later stages in a way I do not understand. I replaced it with 
measureData{iCell} = measureDataTmp{iData}{iCell};
When I tried to plot my sample with this, result looked believable enough to declare it passable, and I hope anything else "fix" broke isn't relevant to my particular data design.

cll008, 2021-07-21:
Sorry I missed this. Looks like the issues were also fixed in a99ae38 and 65aa3d0. If the new code doesn't behave as you expect, do you mind sending a PR so we can compare? Thanks.

closed 2021-07-21 by cll008