sccn/eeglab #301: biosig2eeglab wrong labels channels selection

MultimodalBrainMonitoringUDA, 2021-05-19:
### Description

https://drive.google.com/file/d/1uarj2o3mK0zs8gwgC-f24M6sX7nvKT52/view?usp=sharing
in the above immage is reported the possible bug.
dat.InChanSelect is a vector that goes from 1:number_of_channels but the actual selected channels are in the vector channels.
If you use dat.InChanSelect to select the labels, the labels of the channels don't mach the actual channels. I think the vector channels should be used to select the channels labels.

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2021-06-15:
Sorry for the late answer. The link above says access denied.
Also would you mind to share your files and indicate how to reproduce the error?

MultimodalBrainMonitoringUDA, 2021-06-15:
Sorry! NOw the link should work:
[ https://drive.google.com/file/d/1uarj2o3mK0zs8gwgC-f24M6sX7nvKT52/view?usp=sharing](url)

Atteched the file I tried to open (https://drive.google.com/file/d/1voPiOznPPLuYULkWXO7q7H8Bj1b1qm4m/view?usp=sharing) The problem is that in the plot always the same channels labels are displayed because the variable dat.InChanSelect. Using the varaible channels, the correct labels are selected. I imported the data trought the BIOSIG interface.

arnodelorme, 2021-06-15:
I imported the EDF file with no problem, then I use menu item Edit > Channel location and press OK to look up channel location, removed non scalp channels and plotted spectrum.

arnodelorme, 2021-06-15:
[image]

closed 2021-06-15 by arnodelorme