sccn/eeglab #303: Error when using DIPFIT

math222, 2021-05-29:
### Description

I was able to perform the DIPFIT with the eeglab_dipole.set sample dataset from the tutorial. But when I run the DIPFIT (grid scanning step) on the dataset from https://sccn.ucsd.edu/wiki/BSSComparison (specifically jo74 subject), I get the error message: 
EEGLAB error in function eeglab2fieldtrip() at line 101: struct contents reference from a non-struct array.
Could you help me explain what this message means, and possibly what is the solution?

<br>
#### Steps to Reproduce

1. I suppose this dataset contains location information, and is already filtered.
2. I rund ICA decomposition.
3. I specified the head model.

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     |macOS 10.14.6|
| Matlab version | R2017a |
| EEGLAB version | v2021.0 |

math222, 2021-06-02:
I debug the functions, and find maybe because when "look up" channel locations in the head model setting step, "no location for LEYE,REYE". How do you deal with these two locations?

arnodelorme, 2021-06-02:
WOuld you mind to upload your dataset. @cll008 would you mind to have a look?

math222, 2021-06-03:
Dateset is uploaded to google drive https://drive.google.com/file/d/1o2CHSiVsHi-kPEx_yOFGr8fB5Obl6S0t/view?usp=sharing
which is also available in BSSComparison webpage. I followed the tutorial wiki page to "look up" the "MNI coordinate file for the BEM DIPFIT model", but I guess the "LEYE" and "REYE" channels don't have a location here, so we got an error "non-struct array" (because it's empty).

math222, 2021-06-04:
If the DIPFIT functions indeed require a complete location information of channels, then the issue here is no longer a bug. But I still have two follow-up questions:

1. The BSSComparison dataset is used in several of your publications for ICA analysis. Would you mind sharing your experience of setting up a head model for DIPFIT?

2. Does DIPFIT really require complete location? In the head model setting there is an option to omit channels from dipole fitting. If these omitted channels are not used for DIPFIT (I'm not sure about the details of DIPFIT), is it possible to let DIPFIT function omits their locations, so that the missing of these locations would not be an issue. (In my case here, the error appears because "LEYE" and "REYE" channels' location is empty after looking up a MNI head model.)

arnodelorme, 2021-07-01:
DIPFIT will work without these channel locations. Sorry for the late answer. THere are no accepted locations for these channels.

closed 2021-07-01 by arnodelorme