sccn/eeglab #305: triggers not visible in EEGLAB

jaycbcsgh, 2021-06-04:
We are using gtec machine to collect eeg data. As we are using EEGLAB to analyse data, we added getc extension to import raw data into EEGLAB.
Issue is there are several traiggers in my experiment and after importing data eeglab is not displaying all trigger in event type gui option.

I did not any fix pattern of visibility or non visibility  of trigger. I tried changing trigger value, for example trigger value 9 was not visible but when changes to 90 it was visible. In other, replaced trigger from 120, 121, 122, 123 to 220, 221, 222, 223 then other event values 201, 202, 203 not imported in event value (gui). 

trigger values used in exp
35, 36, 37, 38, 39, 40, 41, 42, 45, 46, 47, 48, 50, 51, 54, 55, 56, 57, 59, 61, 62, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 118, 119, 200, 202, 202, 203..

220, 221, 222, 223

link for raw eeg data

[](https://drive.google.com/drive/folders/1-oWThngtXrtRVEBF8-buNSpnn-j7NHYZ?usp=sharing)

jaycbcsgh, 2021-06-04:
two files 
5 and 6

5 file
trigger value: 70,71,72, 73 not visible

but changed 70,71,72, 73 to 170, 171, 172, 173 then 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115 not visible.

Is there any specific sequence or pattern of numbers required to set a trigger value?

[6.zip](https://github.com/sccn/eeglab/files/6598255/6.zip)

[5.zip](https://github.com/sccn/eeglab/files/6598253/5.zip)

cll008, 2021-06-30:
Hi @jaycbcsgh, does gtec have a native EEG data viewer / processor and are the events all visible there? 
I checked your google drive files (1.mat, 2.mat, 3.mat) as wella s your 5.zip and 6.zip (which are corrupted and also only contain .mat file), should there be some gtec HDF5 files?
I'm not sure if the plug-in is updated or well maintained. Seems that [it is not favorably reviewed](http://sccn.ucsd.edu/eeglab/plugin_uploader/showpluginrating.php?plugin=gtecimport&version=3.15.01).

arnodelorme, 2021-07-06:
The trigger values are not present in the matlab file so there is no way for EEGLAB to extract them.

closed 2021-07-06 by arnodelorme