sccn/eeglab #306: FASTER plugin in EEGLAB needs updating

yanbin-niu, 2021-06-07:
Hello EEGLAB team,

Our team (NDC lab) found that the FASTER plugin in EEGLAB has not been updated to the most up-to-date version of FASTER. 

Specifically, @DMRoberts found that the extension function that gets the list of plug-ins
https://github.com/sccn/eeglab/blob/c57dfeea8cc2dd09c381df4350568cebe9f979be/functions/adminfunc/plugin_getweb.m

Which directs to this list:
https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php

Compared with the most up-to-date FASTER in the Sourceforge page (https://sourceforge.net/projects/faster/), the .zip in the EEGLAB plug-in directory missed:

1. The SourceForge version has the bug corrected in “channel_properties.m”, edited on March 5 2021.
2. The ‘eegplugin_FASTER.m’ file was edited in the EEGLAB version on December 23, 2020.

Can you please help with updating the FASTER plugin in EEGLAB?

arnodelorme, 2021-06-15:
Sorry about the late reply. Before we update could you change the file eegplugin_faster to the reflect current version? (change the version variable). Note that the version online is 1.2.3b so whatever is after that.

Then upload the file to http://sccn.ucsd.edu/eeglab/plugin_uploader/version_update.php

And we will accept the changes right away.

yanbin-niu, 2021-06-15:
Thanks for the reply! I have already uploaded the most up-to-date file for FASTER. One thing to note is that the version number of FASTER didn't seem to be changed when the bug was corrected, both the version in the current EEGLAB and the version on the SourceForge (https://sourceforge.net/projects/faster/) have the same version number of 1.2.3b. Therefore, while uploading, I entered 1.2.3b for the version. Is this going to a problem? Or there is other way to work around this issue?

arnodelorme, 2021-06-15:
Yes, let’s do 1.2.4b

Also when I looked at the eegplugin_FASTER file on sourceforge, it said version 1.0 inside the script.

yanbin-niu, 2021-06-15:
Yes, in eegplugin_FASTER.m, it said `vers = 'FASTER_1.0'`. It also said `FASTER_version='1.2.2b'` in FASTER_GUI.m. My guess is that it did not systematically change this version number over time. 

Anyway, I have reuploaded the file with 1.2.4b. Sorry for the multi-uploading. 

Thank you!!

arnodelorme, 2021-06-15:
OK, thanks.

arnodelorme, 2021-06-16:
I updated the plugin, but there was a bug that prevented it from working. See attached.

arnodelorme, 2021-06-16:
[image]

closed 2021-06-16 by arnodelorme

yanbin-niu, 2021-06-17:
@arnodelorme Thank you very much for looking into this. I wanted to add that:
1. both the old version and new version FASTER throw the same error when calling FASTER with newer versions of Matlab (i.e., Matlab version number R2021a);
2. However, it is worth noting that that the older version of FASTER currently has a critical addition bug;

For these reasons, perhaps the version of FASTER linked in EEGLAB should still be updated, despite the issue you mentioned (which appears to be present in both versions)?  Moreover,  we will communicate this error to FASTER team as well.

mhorger, 2021-11-29:
Hello, I'm receiving the same error pointed out above. Was there a solution to this? Thank you!

arnodelorme, 2021-12-01:
Unfortunately, I cannot fix the issue myself. These are the authors. Maybe try emailing whelanrob@gmail.com 

https://pubmed.ncbi.nlm.nih.gov/20654646/

arnodelorme, 2021-12-01:
Actually, I think it was a typo in the code. I have fixed it and released version 1.2.4 in the plugin manager.