sccn/eeglab #307: Error in pop_chanedit: Reference to non-existent field 'type'.

stevenylai, 2021-06-11:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. Read a cnt file
2. Call pop_chanedit on Neuroscan_64_location.sfp

The equivalent Matlab script is as follows:
```matlab
    EEG = pop_loadcnt('/path/to/file.cnt', 'dataformat', 'auto')
    EEG = pop_chanedit(EEG, 'lookup','/path/to/Neuroscan_64_location.sfp');
```

<br>
#### Expected behavior:
No error

<br>
#### Actual behavior:
Error as follows:

```
readlocs(): 'sfp' format assumed from file extension
Reference to non-existent field 'type'.

Error in pop_chanedit (line 918)
    chans(ind2(index)).type       = tmplocs(ind1(index)).type;

Error in talker_syllable (line 11)
    EEG=pop_chanedit(EEG, 'lookup','/path/to/Neuroscan_64_location.sfp');
```

I am able to fix this by commenting out:
https://github.com/sccn/eeglab/blob/ad8d508265b2deca48e15f9182e34109bfc2cf19/functions/popfunc/pop_chanedit.m#L918 

I can see this line was introduced by https://github.com/sccn/eeglab/pull/276

Is the way I am using Neuroscan_64_location.sfp incorrect?

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 |
| Matlab version | R2017b |
| EEGLAB version | latest develop branch (https://github.com/sccn/eeglab/tree/ad8d508265b2deca48e15f9182e34109bfc2cf19) |

arnodelorme, 2021-06-15:
Would you mind to upload your file Neuroscan_64_location.sfp (compress it as zip and drag and drop here).

It is probably not a real SFP file.

stevenylai, 2021-06-15:
Here it is. The file seems to have the same format as the other sfp files under `sample_locs/`

[Neuroscan_64_location.zip](https://github.com/sccn/eeglab/files/6655031/Neuroscan_64_location.zip)

cll008, 2021-06-30:
Think the SFP file is fine, let me see why the spherical fields don't cause the same error and handle `type` similarly.

PR #338 (2021-07-13): fix 307: chanedit sfp error (missing channel type)
fix #307
prevent error when reading chanlocs files that do not contain type
closed 2021-07-13 by arnodelorme