sccn/eeglab #308: precompute component measures problem

tingyu653, 2021-06-14:
### Description
I used the eeglab STUDY. And when I tried to precompute component measures. 
I checked the ERPs and ERP image. And then I encountered the problem.
[image]

#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*R2020a*] |
| EEGLAB version | [*v2020.0*] |

arnodelorme, 2021-06-15:
This is because you have several datasets for the same subject and they have different ICA decompositions. This is only allowed if you have different sessions (days) for a given subject. Otherwise, if it is a recording on the same day or if the trials are split by condition, the ICA must be the same (you can select both datasets and run ICA on the two datasets).

closed 2021-06-15 by arnodelorme

tingyu653, 2021-06-15:
I change the session of each dataset. But I still encounter the same problem.
[image]

tingyu653, 2021-06-15:
Thanks, the problem is solved now.

arnodelorme, 2021-06-15:
Good happy to know it works now.