sccn/eeglab #310: Runica slowdown

arnodelorme, 2021-06-15:
Just something that I found in the runica() script: if you comment out "drawnow" on line 1047, there is a dramatic speed-up of the script. This has something to do with the GUI and nothing with ICA itself, so I think it could easily be optimized. I'm writing it here because I see that some of the involved people are active on the chat.

arnodelorme, 2021-06-15:
same code are in line 1334

arnodelorme, 2021-06-15:
From Makoto: Wow, the processing time of pop_runica(EEG,'extended',1); changed from 185 s to 72 s after commenting out all the 'drawnow' lines in runica.m. Tjerk is a great hacker! Did you perform time profile analysis or something?

arnodelorme, 2021-06-15:
This is a problem from command line calls as well not only GUIs.

arnodelorme, 2021-06-18:
From Curtis

Replacing ‘drawnow’ with ‘drawnow limitrate’ cuts the processing time in my trials from ~400 seconds to 55-64 seconds. I’m processing 19 channel data, 10 minutes in length, no ERPs. I’m sure the gains will be even more significant with additional channels and especially with multiple data sets within a STUDY.
 
The drawnow function is called in 6 different places in repeating loops. It is designed to allow gui refresh and callbacks to be executed when running within the gui.  Commenting it out removes all interactivity with Matlab or EEGLab while running an extended ICA decomposition. 
 
If you append ‘limitrate’ to the command, it only refreshes the gui every 20fps instead of every time the function is called. If less than 50ms have elapsed since the last call it tosses graphics updates but executes callbacks. Very good interactivity is retained, as is the ability to interrupt the ICA decomposition.
 
Merely commenting out the drawnow doesn’t materially change the execution time on my data. However, as a matter of good coding practice this should be turned off altogether if not running inside a GUI.
 
I recommend that, at the very least, runica.m be modified to replace ‘drawnow’ with ‘drawnow limitrate’.

arnodelorme, 2021-07-06:
@cll008 try solution above (‘drawnow limitrate’ )

Allexxann, 2021-07-11:
I was simply commenting out the whole thing for several years.

commit 7bbef00e (2021-07-16): Update runica.m

fix #310. EEGLAB sample_data runica takes 32 instead of 81 seconds with `limitrate` option
closed 2021-07-16 by arnodelorme