sccn/eeglab #318: Error with channel location using .elc file StandardWaveguard64 from ASA lab to EEGlab

sasis001, 2021-06-19:
### Description
Hello,

I am receiving an error while trying to use channel locations to import a .elc file from ANT 
Neuro after reading the CNT files using the EEProbe plugin. I have tried to open the different channel location files and receiving the same error message with all the files. 

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce
Here is what I have been doing.
1. Saved a .elc file from ANT Neuro.
2. Tried to obtain Channel locations from EEGLAB by selecting "Read locations" and the .elc file. Receiving the attached error message.
3. Tried selecting a standard Look up locs (BESA, MNI, spherical), same error message.

<br>
#### Expected behavior:
I should have received a printout of the channel locations.

<br>
#### Actual behavior:
Error message attached.

<br>

|----| ---- | 
| OS version     | [Windows 10]|
| Matlab version | [*2021*] |
| EEGLAB version | [*2021*] |
[image]

Jay

arnodelorme, 2021-06-21:
Hi Jay, Would you mind to share a data file and an electrode file?

sasis001, 2021-06-22:
Hello,

I was able to import the channel locations. 

I did have another question though. I am trying to process files that have the epochs of interest randomly interspersed within a file. Each subject has several files and each file has 3-4 trials corresponding to each event of interest. In viewing your workshop on processing with EEGLAB III, it seems I might have to epoch by event for each file per subject and then eventually analyze all of the files via the study design module in EEGLAB. Is this the correct approach to analyzing trials that corresponding to several events of interest.

For instance, in each file per subject I have the 'yes' and 'no' responses randomized, and additionally the yes responses have to be categorized by certain criteria that are distinguished by EEG marker numbers within a set. Can I separate these different trials types when obtaining the epochs or is there a different format for this that you might recommend?

Thanks,

Jay

sasis001, 2021-06-24:
Hello,

Could you confirm that the channel locations indicated

closed 2021-06-24 by sasis001