sccn/eeglab #321: Variable "CURRENTSTUDY" is not set to 0 when calling "eeglab redraw" from a empty MATLAB workspace

CloudyDory, 2021-06-24:
### Description

If we start EEGLAB by command `eeglab`, the variable `CURRENTSTUDY` is set to `0`, but if we start EEGLAB by `eeglab redraw`, the variable is set to `[]`. If we then load more then two ".set" files, the value of `CURRENTSTUDY` is still `[]`, and we cannot switch dataset by clicking on "Datasets" in the main menu. This will cause an error: 

`
Operands to the logical and (&&) and or (||) operators must be convertible to logical scalar values. 
Error while evaluating Menu Callback.
`

Manually set `CURRENTSTUDY` to `0` before switching dataset can fix this error, but ideally the correct value of `CURRENTSTUDY` should be set automatically. It seems that the cause of the problem is that the following lines (lines 373-386 in `eeglab.m`) are not executed when running `eeglab redraw`:

```
if nargin < 1 || exist('EEG') ~= 1
	EEG = [];
    ALLEEG = [];
    CURRENTSET = [];
    ALLCOM = [];
    LASTCOM = [];
    STUDY =[];
    CURRENTSTUDY = 0;
	EEG = eeg_emptyset;
	eegh('[ALLEEG EEG CURRENTSET ALLCOM] = eeglab;');
    if ismatlab && get(0, 'screendepth') <= 8
        disp('Warning: screen color depth too low, some colors will be inaccurate in time-frequency plots');
    end
end
```

<br>
#### Steps to Reproduce

1. run `clear all` in MATLAB command window.
2. run `eeglab redraw`. Notice that now the value of `CURRENTSTUDY` is `[]`.
3. Click on `File -> Load existing dataset`, select `eeglab_data.set` and `eeglab_data_epochs_ica.set` in the "sample_data" folder. It will load the two selected data files.
4. Click on "Datasets" in the main menu and select dataset 1. This triggers the above error.

<br>
#### Expected behavior:

EEGLAB successfully switch the current dataset.

<br>
#### Actual behavior:

EEGLAB cannot switch the current dataset, an error is shown:

`
Operands to the logical and (&&) and or (||) operators must be convertible to logical scalar values. 
Error while evaluating Menu Callback.
`

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 21H1 |
| Matlab version | MATLAB R2021a Update 3 |
| EEGLAB version | 2021,0 |

arnodelorme, 2021-06-24:
I could not reproduce the problem. Are you using the dev version of EEGLAB? I have added a line to EEGLAB

closed 2021-06-24 by arnodelorme

CloudyDory, 2021-06-25:
No, I'm not using the dev version. I just downloaded the "eeglab_current.zip" from [ftp://sccn.ucsd.edu/pub/daily/](url) and I can reproduce the problem.

arnodelorme, 2021-06-25:
OK, thanks for letting us know. This will be fixed in the next release next month then.