sccn/eeglab #322: Modulation Index Value EEGLAB PACTools

cristinabenetton, 2021-06-28:
### Modulation Index Value EEGLAB PACTools

I have one question regarding the plugin PACTools to compute PAC on EEG resting state data. 

After estimating PAC on my dataset, I plot the comodulogram results. Do you know if there is a way in which I can extract the Modulation index single value to then do statistics on it? Either the maximum modulation index value in the comodulogram or maybe a z-score based on the comodulogram.

<br>
#### Steps to Reproduce

1. Open a dataset in EEGLAB
2. Compute PAC [_Tools --> PAC Tools --> Estimate PAC_]

[image]

3. Plot PAC estimation [Tools --> PAC Tools --> Plot PAC]

[image]

4. Plot the Comodulogram

Here I stop and don't know where to find the Modulation Index value

arnodelorme, 2021-07-01:
@nucleuscub any clue? Tx.

arnodelorme, 2021-07-06:
@nucleuscub how can we help?

nucleuscub, 2021-07-07:
Hi Arno, I transferred this issue to the pactools repo and replied to her
from there. Alll good with this. Ppl just don't read the documentation. 🙄

Thks for letting me know.
Cheers.
R

arnodelorme, 2021-07-07:
Thanks so ugh Ramon for taking the time.

closed 2021-07-14 by cll008