sccn/eeglab #334: Work on Matlab script to generate data for deep learning

arnodelorme, 2021-06-30:


arnodelorme, 2021-07-15:
1. You were able to run restingstate_prepare_clean_master.m

2. Modify it to run on all data. All data which is in **child-mind-uncompressed**

3. Run python VGG16 program, SexPrediction-VGG-Topo.ipynb
(will need to run on Expanse)
See Dung demo to start Jupyter notebook on Expanse

4. Adapt the script https://github.com/arnodelorme/child_mind_2020/blob/master/vgg16/dl_vgg16.m to take the same input data. Run on expanse as well.

arnodelorme, 2021-07-15:
@jimatbrock see notes above

arnodelorme, 2021-07-15:
This is the video

https://drive.google.com/file/d/168-f5BASXbcz--FvP2gKlYZT4uU4ke4S/view?usp=sharing

jnnstchr, 2021-07-16:
Am I supposed to run the script on the whole data after all, or just make sure it would work for all the data? I remember we were talking about two options and I'm not sure which one we picked.

jnnstchr, 2021-07-16:
@dungscout96 could you send me the list of dependencies you're using in your ml environment?

dungscout96, 2021-07-16:
You only need to check to make sure it would work. Don't need to run on all data. I pushed the environment file here: https://github.com/sccn/childmind/blob/master/environment.yml

jnnstchr, 2021-07-17:
I cannot set up my anaconda environment properly, I tried running conda env create -f environment.yml to base it on the above yml file, however I get:
 ResolvePackageNotFound: 
  - cvxpy
  - torchtext=0.9
  - pytorch=1.8.0
  - stanza.
  Then I decided to install the dependencies manually, which works for some of them, but for instance for pytorch I get:
EnvironmentNotWritableError: The current user does not have write permissions to the target environment.
which likely means I need to run it as a root in expanse. Any hints on how could I do it/how to avoid it?

dungscout96, 2021-07-17:
Did you create a new environment and install there or you were on the _base_ env? Maybe you can send the list of commands up until the one that failed. I can share that with Expanse team to see what the issue could be too.

jnnstchr, 2021-07-19:
I created a new environment, but I _might_ have not activated it, I decided to start from scratch and create a new one. The commands I am using right now are:
module load anaconda3
conda create --name ML
conda activate ML

and now the issue:

> CommandNotFoundError: Your shell has not been properly configured to use 'conda activate'.
> To initialize your shell, run
> 
>     $ conda init <SHELL_NAME>
> 
> Currently supported shells are:
>   - bash
>   - fish
>   - tcsh
>   - xonsh
>   - zsh
>   - powershell
> 

The abovementioned commands (e.g.  conda init bash) result in:

> We trust you have received the usual lecture from the local System
> Administrator. It usually boils down to these three things:
> 
>     #1) Respect the privacy of others.
>     #2) Think before you type.
>     #3) With great power comes great responsibility.
> 
> PIN+Yubi: 
> 
which from what I understand requires me to have permissions that I don't have right now. 
Thanks for your help!

dungscout96, 2021-07-19:
I vaguely remember this issue.. Can you try 
```
vim ~/.bashrc
```
then paste this at the bottom of it
```
# >>> conda initialize >>>
# !! Contents within this block are managed by 'conda init' !!
__conda_setup="$('/cm/shared/apps/spack/gpu/opt/spack/linux-centos8-skylake_avx512/gcc-8.3.1/anaconda3-2020.11-bsn4npoxyw7jzz7fajncek3bvdoaa5wv/bin/conda' 'shell.bash' 'hook' 2> /dev/null)"
if [ $? -eq 0 ]; then
    eval "$__conda_setup"
else
    if [ -f "/cm/shared/apps/spack/gpu/opt/spack/linux-centos8-skylake_avx512/gcc-8.3.1/anaconda3-2020.11-bsn4npoxyw7jzz7fajncek3bvdoaa5wv/etc/profile.d/conda.sh" ]; then
        . "/cm/shared/apps/spack/gpu/opt/spack/linux-centos8-skylake_avx512/gcc-8.3.1/anaconda3-2020.11-bsn4npoxyw7jzz7fajncek3bvdoaa5wv/etc/profile.d/conda.sh"
    else
        export PATH="/cm/shared/apps/spack/gpu/opt/spack/linux-centos8-skylake_avx512/gcc-8.3.1/anaconda3-2020.11-bsn4npoxyw7jzz7fajncek3bvdoaa5wv/bin:$PATH"
    fi
fi
unset __conda_setup
# <<< conda initialize <<<
```
Save the changes then `source ~/.bashrc` or open a new terminal and try `conda activate` again

jnnstchr, 2021-07-20:
Hi @arnodelorme, in your dl_vgg16.m script you're importing vgg16.mat file, which according to README should be in vgg16 directory. It looks like it's missing - could you push it to the github? Thank you!

closed 2021-08-16 by cll008