sccn/eeglab #336: Acquire login on XSEDE and SDSC

arnodelorme, 2021-06-30:


arnodelorme, 2021-07-08:
See video @jnnstchr

https://drive.google.com/file/d/1RtCOqC3DOOt0UJDbGoJXHfhGsX1KDvcJ/view?usp=sharing

arnodelorme, 2021-07-08:
Tasks:
- Get the script to run on Expanse
- Try to do commits on our repo, set up SSH for Github

arnodelorme, 2021-07-08:
In addition to setting up SSH with expanse, you might want to use the same private key for GitHub. You would need to add both your local key and your expanse key. This way you can push code from both sides.
Here is how to do it

https://www.youtube.com/watch?v=nQDFBd5NFA8

jnnstchr, 2021-07-15:
I did setup the SSH keys both my local computer and expanse, created my own branch for childmind repository and pushed the modified script, pulled the master branch on my expanse account. I do not have the permission to access /expanse/projects/nemar, however I managed to copy both the eeglab and the previously used file from my local computer with sftp, unpack it, modify the path again and run it on expanse.

arnodelorme, 2021-07-15:
This part is done.

closed 2021-07-15 by arnodelorme

jnnstchr, 2021-07-16:
I'm added to nemar group now and all the paths work as expected

arnodelorme, 2021-07-16:
Great.