sccn/eeglab #337: Missing channels in plot (scroll) view

baniknan, 2021-06-30:
Hi! We're running some pilot EEG sessions with BrainVision Recorder and  we no longer see all the channels when we use: Plot -> Channel data (scroll). I have tried increasing the scale (to around 400), and then I'm able to see all the channels but they're pretty much a straight line. 

I'm so confused if we're seeing incorrect data or increasing scale till you see all the channels is a common practice? Our previous datasets did not have this issue. 

Not sure if this is a Brain Vision recorder issue or EEGlab issue. Thanks!

cll008, 2021-06-30:
Hi @baniknan how did the data look when you were recording it with the BrainVision Recorder (there is a live view of the incoming data right?)
Did you recently update EEGLAB? Try importing one of your previous known-to-be-good datasets and see if the issue persists. If not, maybe your latest problematic dataset is corrupted.

cll008, 2021-06-30:
Also, when viewing the EEG scroll plot, did you select `Remove DC Offset` from the display menu before increasing the scale?

closed 2021-07-13 by arnodelorme

baniknan, 2021-07-14:
Hi there!

My apologies for the delayed response. The data looked good while recording, only afterwards did we see a problem.

We have an old dataset that was converted from BV file to .EDF and that shows up well on the scroll. So, I wasn't sure if it's a BV plug-in issue or an issue with how the data was recorded.

What are some possible reasons a data file can become corrupted? We haven't done anything to it except convert it using BV converter and import it to MATLAB.

I'll check if we have the latest EEGLab on MATLab. We should have the latest one because I installed it only a couple of months ago.

Best,
Nandita

cll008, 2021-07-14:
Hi Nandita please also check point 2 in my previous comment.
It's difficult to guess what's wrong with one dataset. Usually it's not an issue with how the data is recorded, and we haven't had any other BV plug-in issues (it's a popular format).