sccn/eeglab #339: RDF format problem

arnodelorme, 2021-07-01:
Can you try your hand at importing a subject for me into the Sternberg database. Aside from this one I am close to a re-imported (and so not filtered) version for all. The events for half are a sticking point, and for this one the data itself can't be imported. 

The data:
/data/cta/SternbergDatabase/subj13_Stern1.RDF
/data/cta/SternbergDatabase/subj13_Stern2.RDF
/data/cta/SternbergDatabase/subj13_Stern3.RDF
/data/cta/SternbergDatabase/subj13_Stern4.RDF

I think I originally used (and use for the remainder of the RDF subjects) this function:
  EEG = pop_read_erpss([newbase,'subj',int2str(nx),'_Stern',int2str(ds),'.RDF']);

It takes a while, then imports data of size:
data: [18432×0 double]

which is of course wrong. 

For some reason it also sees the srate as:
srate: 16.8364

This subject may have been special given that he is the first in the RDF folks. I have no idea how/why it was different. 

If you can't figure it out, I guess we'll just leave him with the .set instead of the raw data and make a note.

arnodelorme, 2021-07-01:
@cll008 would you mind to have a look

cll008, 2021-07-09:
I tried importing with the Biosig interface (which works for the other .RDF) as well and subj13's files error on fopen. RDFs don't seem editable in txt or excel so I don't have a good way of inspecting further. Maybe the raw files were corrupted?

closed 2021-07-09 by cll008