sccn/eeglab #340: replace ft_inside_vol with ft_inside_headmodel

robertoostenveld, 2021-07-02:
the function ft_inside_vol has been replaced already quite some time ago with ft_inside_headmodel. However, I see that eeglab https://github.com/sccn/eeglab/blob/develop/functions/sigprocfunc/dipoledensity.m still uses find_inside_vol, which is even the predecessor of ft_inside_vol. 

right now dipoledensity has a subfunction find_inside_vol, which calls ft_inside_vol. If that call were replaced by ft_inside_headmodel, it would already be sufficient. However, better would be update the code altogether.

arnodelorme, 2021-07-02:
Thanks Robert, @cll008 would you mind to investigate?

cll008, 2021-07-09:
Sure. Looks like the deprecated fxn is used in [std_dipoleDensityStatistics](https://github.com/sccn/std_dipoleDensity/blob/master/std_dipoleDensityStatistics.m) as well and I update those. 

@arnodelorme Does EEGLAB look for these functions in the Fieldtrip-lite plug-in? Are there other private copies I need to worry about?

PR #348 (2021-07-16): Update dipoledensity.m and runica.m
fix #340. `Outside` might be unnecessary (only used here to save a file) but keeping for backwards compatibility.
closed 2021-07-16 by arnodelorme