sccn/eeglab #342: When using pop_chanedit, an error occurs even if "use MNI coordinate file for BEM dipfit model" is selected.

iseda623, 2021-07-06:
### Description
When running the following, an error occurs and the channel cannot be set
https://eeglab.org/tutorials/04_Import/Channel_Locations.html#:~:text=pop_chanedit(%20chanlocs%20)%3B

`
>> chanlocs = struct('labels', { 'cz' 'c3' 'c4' 'pz' 'p3' 'p4' 'fz' 'f3' 'f4'});
pop_chanedit( chanlocs );
readlocs(): 'chanedit' format assumed from file extension
Reading file (lines): 10 20 30 40 50 60 70 80 90 100 110 120 130 140 150 160 170 180 190 200 210 220 230 240 250 260 270 280 290 300 310 320 330 340 350 358
readlocs() warning: Fewer columns in the input than expected.
                    See >> help readlocs
Warning: coordinate conversion failed
フィールド名 "Z" が認識されません。
エラー: pop_chanedit (行 911)
                        chans(ind2(index)).Z          = tmplocs(ind1(index)).Z;
エラー: pop_chanedit (行 268)
        [chans chaninfo urchans com] = pop_chanedit(chans, chaninfo, 'lookupgui', []);
`

<br>
#### Steps to Reproduce

1.Pass the eeglab tutorial to the following
https://eeglab.org/tutorials/04_Import/Channel_Locations.html#:~:text=pop_chanedit(%20chanlocs%20)%3B
2. "use MNI coordinate file for BEM dipfit model" is selected.
3. Select OK.

<br>
#### Expected behavior:

pop_channeledit should open and display the value.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*R2021a*] |
| EEGLAB version | [*2021.0*] |
Download from the following link | https://sccn.ucsd.edu/eeglab/currentversion/eeglab_current.zip

arnodelorme, 2021-07-06:
Did you update to Dipfit 4.1
This fixes the problem.

iseda623, 2021-07-06:
I updated to 4.1 and that solved the problem.
Thank you very much.

closed 2021-07-06 by iseda623