sccn/eeglab #346: Error in function checkoldformat() when opening datasets with ICA weights

jmkendallbar, 2021-07-08:
Hello, 

I ran ICA on a subset of my data and was able to save that dataset (and re-open it later). When I import an EDF file using the BIOSIG toolbox and then apply the ICA weights of the previous dataset to this dataset (Edit Dataset> ICA weights (from other dataset)), I save the dataset and receive this error if I try to re-open the saved dataset. The datasets are quite big (>1GB) and I am wondering if that is what is causing this issue. Here is the error message when I try to reopen the new dataset with the transferred ICA weights:

[image]

Please let me know if someone can assist me, or if this question is better directed elsewhere. Thank you.

cll008, 2021-07-14:
Hi @jmkendallbar, which version of EEGLAB are you using? 
It's unlikely that your issue is because of file size. The error you mentioned was at line 277 but should be this line on the latest develop version of EEGLAB:
https://github.com/sccn/eeglab/blob/a99ae389f573f38cadfe7602d7ece837f1bc777d/functions/popfunc/pop_loadset.m#L284

Let's give your steps some names...

1. Data with ICA
2. Another EDF file imported
3. Apply ICweights from 1 onto 2
4. Save 3
5. Load 4

Can you check `isempty(EEG.data)` after step 3? Otherwise, it might be easier for you to just share with me your data from 1 and 2 and I can try to debug it on my end.

arnodelorme, 2021-07-16:
Would you mind sharing the big dataset (Dropbox etc...).
Thanks

arnodelorme, 2021-07-19:
Please reopen the bug as needed.

jmkendallbar, 2021-07-19:
Hello Arnaud & Clement,

Thank you for your help. Sorry for the delay, I was waiting for these
datasets to upload. Please request GoogleDrive access for the following
files and I will give them to you.

Here are links to the datasets:

Dataset 1: excerpt of EDF file with generated ICA Weights:
.set file:
https://drive.google.com/open?id=1zFhFOkFCaMHC6RHGJslNncLnbN4Sye9V&authuser=jmkendal%40ucsc.edu&usp=drive_fs
.fdt file:
https://drive.google.com/open?id=1gDkXWZf65AszkK19Wh3FCUv6B-DfnFj6&authuser=jmkendal%40ucsc.edu&usp=drive_fs

Dataset 2: whole EDF file with ICA Weights transferred over:
https://drive.google.com/open?id=10SZ4YMwW-vpEXOxUhmxUsGbvoIcTAihS&authuser=jmkendal%40ucsc.edu&usp=drive_fs

EDF file (2.76 GB - only import channels 1-9 with electrophysiological
data):
https://drive.google.com/open?id=18z7uoVCYjbTO_akEE6V2sQOwwDw-DHc8&authuser=jmkendal%40ucsc.edu&usp=drive_fs

I was able to do a work-around where I generated a new variable and stored
EEG.data and EEG.icaact from original and pruned datasets, then load the
new Matlab variable in as a new dataset, and write the new EDF file.
However, this takes up a lot of memory and the writeeeg step is very slow.
I am searching for a more elegant solution to combine raw EEG, pruned EEG
with ICA, and ICA activations, into a single EDF file. It is also helpful
for me to be able to save intermediate steps, but I haven't been able to
open the dataset #2 listed above. I noticed that EEGLAB did not save an
.fdt file for this dataset. Could that be the issue?

Thank you for any help you can provide!

Jessie Kendall-Bar

-- 
Jessica Kendall-Bar (she/her)
NSF Graduate Research Fellow
Department of Ecology and Evolutionary Biology
University of California, Santa Cruz
Williams <https://williams.eeb.ucsc.edu/lab-members/jessica-kendall-bar/>
& Costa
Lab <https://costa.eeb.ucsc.edu/current-students/jessica-kendall-bar/>
***@***.***
jessiekb.com

cll008, 2021-07-20:
Just requested access, will take a look when able.

EEGLAB automatically saves the `.fdt` when saving `.set` (based on EEGLAB GUI > File > Preferences). The default for 2021 onwards is to save only 1 file, but older versions default to saving `.set` and `.fdt`.

Additional info that could be helpful here, what OS are you using and which version of EEGLAB and MATLAB? Thanks.

jmkendallbar, 2021-07-20:
Just provided access, hopefully that works for you.

I am running Windows 10 with EEGLAB version 2021.0 and Matlab R2020b

cll008, 2021-07-20:
I confirm receiving the same error when I load your ICA weights applied set file.

However applying the weights and saving the dataset worked for me. When you are applying ICA weights from dataset 1 (with ICA weights), does your menu look like this (I imported the first 9 channels as you said and only the first 1000 timepoints):
[image]

And then from there you click okay then use `File > Save current dataset as`? and the output is the 'bad' file? (I also changed my preferences to test saving just `.set` and also saving both `.set` and `.fdt` and I could load both output files). I'll test it with the full EDF file now, maybe the data length and size causes some unexpected interaction.

jmkendallbar, 2021-07-20:
Yes, this is exactly how I did it. When re-opening the full dataset with the transferred ICA weights (using the same method), I get that error.

Glad that you are able to recreate it so far!

cll008, 2021-07-20:
So I get this message when trying to save the full file as a single `.set` file.
[image]

Changing this setting to save `.set` and `.fdt` separately results in a working file.
[image]

Am I reading this correctly that your file is about 4 days in runtime? Working with a file this long will take a long time in saving and loading. I suggest recording separate datasets if possible, and barring that, downsample early on in preprocessing (if applicable).

closed 2021-07-20 by cll008