sccn/eeglab #347: Sampling rate bug with EDF Files

skoenig3, 2021-07-09:
Hello,

I found a bug in the BioSig 3.7.5 plugin for reading EDF data with annotations. I realize this is not a problem with eegLab but one if it's commonly used plugins. I figured it would report it since other people may run into similar issues and it took me a while to find this bug. This bug also appears unresolved in BioSig 3.7.8 too. 

Function sopen.m, Lines 639-643
            for k=chan,
                if (HDR.AS.SPR(k)>0)
                    HDR.SPR = lcm(HDR.SPR,HDR.AS.SPR(k));
                end;
            end;
            
For me, HDR.AS.SPR has the correct sampling rate of 500 Hz for our the EEG channels and for whatever reason a sampling rate of 57 Hz for the EDF annotations (I've never found these to be accurate). Since the code does not ignore the EDF Annotations sampling rate, the least common multiple (LCM) of 500 and 57 is 28,500!  Our EDF files without annotations have no problem being read in correctly. 

Thank you,
Seth Koenig

arnodelorme, 2021-07-13:
@amisepa would you mind to answer?

amisepa, 2021-07-13:
Hi Seth,

Thank you for pointing it out. Yes there are many issues with EDF annotations because of the high diversity in recording formats. It is not easy to cover all the possibilities with the plugin. 

Have you tried the mexSLOAD option during importation (a box to check in GUI in import window)?

You can also try this new function I am working on, but haven't run enough tests yet to make it available in eeglab.
[import_EDF.zip](https://github.com/sccn/eeglab/files/6811760/import_EDF.zip)

Add it to your path, launch eeglab, and type `EEG = import_EDF();`

Let me know if that works for you.

Cedric

skoenig3, 2021-07-14:
Hello Cedric,

Thanks for the suggestion. I temporarily patched the code myself already
but I will try these out too. The easiest solution was just to ignore the
annotations when determining the sampling frequency.

Seth

arnodelorme, 2021-07-16:
Has this been resolved? I am closing the bug report for now, but feel free to reopen.

closed 2021-07-16 by arnodelorme