sccn/eeglab #352: MW project: modify preprocess data and participants

arnodelorme, 2021-07-15:


arnodelorme, 2021-07-15:
@dungscout96 we had a list of items to modify on the data preparation. What I remember and picked up from the paper.

- Make sure people of the same age for each gender. Maybe create a script to balance participants based on the label used for classification.

- Make sure the exact same number in each class (there was a difference of 1 if I remember well between Males and Females)

- Milham: the HBN is a clinical sample...males more likely to have ADHD, females to have anxiety/mood. probably not relevant for this paper, but something to be aware of. Also, males tend to have high motion than females...any chance this is what is being picked up? - I realize this is an fMRI imagers concern

- Low pass filtering below 25 Hz. Not needed. We should use the NIQUIST at 64 Hz

- High pass filtering. Probably 0.01 Hz better than 0.25 Hz. Let's talk.

- For spectral decompostion, check overlap (50 in the paper but 50% would be 64 samples)

- Image pixel rescaled for each subject (for scalp topo). We loose absolute amplitude information. Maybe better to rescale uniformly across all participants.

arnodelorme, 2021-07-22:
Summary of call 7/22

- Joanna to check we have all subjects in /expanse/projects/nemar/child-mind-uncompressed (compare to http://fcon_1000.projects.nitrc.org/indi/cmi_healthy_brain_network/sharing_neuro.html)

- Joanna to run scripts https://github.com/sccn/childmind/blob/master/restingstate_prepare_clean_master.m on all subject and document subjects which fail

- Joanna to check how to package large datasets for Matlab Deep Learning. Joanna to check if there are solution to reuse Pytorch datastore from Matlab.

- Jung and Arno to modify restingstate_prepare_clean_master.m to generate output matrices that can be readily used for Deep learning.

- Joanna to continue investigate vgg16 and Matlab on Notebook https://github.com/arnodelorme/child_mind_2020/blob/master/vgg16/dl_vgg16.m (the vgg16.mat is in /expanse/projects/nemar/child_mind_2020/vgg16/vgg16.mat and is generated by saving the "net" variable under a workstation that has the vgg16 add-on installed).

dungscout96, 2021-07-23:
Preprocessing script has been updated and also is saving the data matrix for eye closed, as well as subject IDs associated with each sample, at the end of the script. I saved it into a separate file here to keep the old one for legacy purpose: https://github.com/sccn/childmind/blob/master/restingstate_prepare_clean_package.m

arnodelorme, 2021-07-27:
Message from Vijay (for future reference)

As you’ve likely seen: the ONNX importer can be used to import pretrained pyTorch models, i.e. if you’re not able to solve the “data feeding” problem you’re currently facing in MATLAB

https://www.mathworks.com/help/deeplearning/ref/importonnxnetwork.html

jnnstchr, 2021-07-31:
Message from Vijay (also for future reference)

With some processing in Python the data can be saved as a ‘mat’ file.

 

Python Code:

from torchvision import datasets

from torchvision.transforms import ToTensor

import scipy.io

 

training_data = datasets.FashionMNIST(

    root = "data",

    train = True,

    download = True,

    transform = ToTensor()

    )

 

x = training_data.train_data.numpy()

y = training_data.train_labels.numpy()

 

scipy.io.savemat('test.mat',dict(data = x,label = y))

 

 

“training_data” is the PyTorch Dataset object.

“train_data” and “train_labels” are members of the object “training_data”. There are other members like “test_data”, “test_labels”, etc.,

Convert them to numpy and save into a MAT file.

 Message from Vijay (also for future reference)
With some processing in Python the data can be saved as a ‘mat’ file.

 

Python Code:

from torchvision import datasets

from torchvision.transforms import ToTensor

import scipy.io

 

training_data = datasets.FashionMNIST(

    root = "data",

    train = True,

    download = True,

    transform = ToTensor()

    )

 

x = training_data.train_data.numpy()

y = training_data.train_labels.numpy()

 

scipy.io.savemat('test.mat',dict(data = x,label = y))

 

 

“training_data” is the PyTorch Dataset object.

“train_data” and “train_labels” are members of the object “training_data”. There are other members like “test_data”, “test_labels”, etc.,

Convert them to numpy and save into a MAT file.

 

This data can further be loaded into a Datastore to apply transforms/create partitions … https://www.mathworks.com/help/matlab/datastore.html

closed 2021-08-16 by cll008