sccn/eeglab #353: pop_selects returns the wrong set of channels with 'nochannel' option

crisglav, 2021-07-16:
### Description

`outEEG = pop_select(EEG,'nochnnel',badchannels)` returns a different set of removed (bad) channels in `outEEG.chaninfo.removedchans`

<br>
#### Steps to Reproduce
Using a 64 channels EEG recording,
```
badchannels = [1 17 18 37];
outEEG = pop_select(EEG,'nochnnel',badchannels);
```

<br>
#### Expected behavior:
I would expect to have `EEG.chanlocs(badchannels)` in `outEEG.chaninfo.removedchans`

<br>
#### Actual behavior:
I actually have `EEG.chanlocs([badchannels 2 20 21 41]) ` in `outEEG.chaninfo.removedchans`

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Ubuntu 18.04.5] |
| Matlab version | [R2020b] |
| EEGLAB version | [v2021.0] |

arnodelorme, 2021-07-16:
This was fixed recently. Have you tried the latest version on Github?

closed 2021-07-16 by arnodelorme

crisglav, 2021-07-21:
Dear Arno,

I have checked the last version of the develop branch and the bug is still there (pop_select.m lines 563-589). Which version should I use?

Best, 
Cristina