sccn/eeglab #354: error in function mcolon() at line82

benoalbe, 2021-07-17:
### Description
I'm using eeglab2021, Matlab 2020a and when I try to plot channel measures it appears this error when I'm going to run more two channels 
[image]

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

cll008, 2021-07-19:
Hi there, could you be more specific about how I would reproduce the bug? For example, which measures are you trying to plot? What OS are you on?

Looks like `mcolon` may have to be compiled by the user from [here](https://www.mathworks.com/matlabcentral/fileexchange/29854-multiple-colon). Not sure. I'll tag @MakotoMiyakoshi here since ARfitStudio also uses mcolon, maybe he has an idea.

MakotoMiyakoshi, 2021-07-19:
I'm surprised to learn that EEGLAB uses mcolon... where exactly is it used?

cll008, 2021-07-19:
Not sure, let's see if @benoalbe has more info on which study measure he was plotting...

arnodelorme, 2021-07-19:
Hi Makoto, yes, this is an ARfitStudio problem (not a study problem). The user does not mention using a study.

closed 2021-07-21 by cll008