sccn/eeglab #355: removing one IC from one of clusters

tingyu653, 2021-07-18:
### Description
After clustering, I removed one of IC from one cluster. After that, I tried to see the result after removing. But I got several errors from Matlab.
[image]
The result also doesn't seem right. I found out that the IC I removed still exist, and the of the IC changed into s03A/IC15.
[image]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*R2020a*] |
| EEGLAB version | [*v2020.0*] |

cll008, 2021-07-19:
Could you please elaborate on how you removed the IC, and what process you are using to 'see the result'? Thanks.

arnodelorme, 2021-07-19:
Would you mind sharing your STUDY, and providing a detailed description on how to reproduce the problem?

tingyu653, 2021-07-20:
I pressed the "Reassign selected component(s) " button, and removed the IC from Cls 7 to outlier. To 'see the result', I pressed the 'Plot ERP(s)' button at right column.(Cls 7->all components) immediately.
[image]
After I posted the question, I tried not to press the 'Plot ERP(s)' button immediately, and I saw the correct ERP(s). 
But I still wonder why couldn't I press the 'Plot ERP(s)' button immediately?

cll008, 2021-07-20:
I see. Glad you figured it out.
It just takes time to shuffle and reindex the data. We could freeze the GUI or MATLAB when this process is happening but I think we would have more problems doing that.

closed 2021-07-20 by cll008