sccn/eeglab #356: Exporting Fieldtrip data to EEGLAB

NirOfir, 2021-07-22:
### Description

Hi, I'm working on an EEG dataset on FT, and I wanted to use LIMO to analyze a part of it. Since FT to LIMO interface is not yet clear, I tried exporting the preprocessed and epoched data to EEGLAB, and using LIMO from it. Then I came across the issue of using fiedltrip2eeglab(), which relies on ft_read_data(). From what I understand, this function is used to read data from the disk in the format of whichever system was used to acquire it into an FT struct, which is not really what I was trying to do. Is there a way currently in eeglab to export an FT struct (something like what ft_preprocessing would return) into eeglab?

| OS version     | CentOS 7 |
| Matlab version | 2018b |
| EEGLAB version | 2021.0 |

NirOfir, 2021-07-25:
OK, I've been reading deeper into the code in pop_fileio and I think I understand it better now. Let me know if I got it right. 

In order to import an existing data structure from fieldtrip, we need the header (the output of ft_read_header), a raw data matrix (channels X timepoints X trials), and optionally the events (a string to feed into ft_read_event or the output of the same function).

1. Is fieldtrip2eeglab intended to work only for "raw" data structures (according to ft_datatype_raw)?
2. Could we circumvent the reliance on the header? I can see scenarios in which it will be problematic. For example, if I acquired my data with a 512 Hz sampling frequency, but downsampled it as part of the preprocessing, eeglab would consider this data to have the false original Fs.
3. Would it be better to collect the data matrix within fieldtrip2eeglab from the FT data structure rather than having the user do it themselves before calling this function?

I would be happy to make a PR with a more detailed help section and any other changes necessary.

CPernet, 2021-07-26:
@arnodelorme can you help?

arnodelorme, 2021-07-26:
You can recreate the Fieldtrip header structure from the Matlab command line. Look at pop_fileio for what fields are needed in the header. Please submit a report of your efforts here so we can share it with others.

Otherwise, you can also import raw data matrices (variables) using the EEGLAB function pop_imporset.m

closed 2021-07-26 by arnodelorme

NirOfir, 2021-07-27:
Thanks for your response. I'll get back to you soon with a more complete code that takes an FT data structure as input, creates the necessary inputs for pop_fileio, and calls it.

NirOfir, 2021-08-01:
I'm writing this issues here, so I'll remember them it when I get around to this issue:
1. If channels are interpolated in FT, The order of the channels is broken (interpolated channels are concatenated at the end). This seems to be an issue for EEGLAB, at least when using LIMO.
2. There are a whole bunch of fields in EEGLAB that are never created in FT (setnames and such). I think it will be nice to have a "fields" input as a key-value cell array, so users can fill those in.