sccn/eeglab #358: std_limo sets default method to WSL instead of WLS

NirOfir, 2021-07-27:
### Description
A small typo in std_limo() line 104 sets the GLM solution method to WSL instead of WLS by default.

Should I make a PR to change
`opt.method         = 'WSL';`
to
`opt.method         = 'WLS';`
?

Thanks!

NirOfir, 2021-07-27:
Another thing that I noticed now: std_limo checks in lines 286-296 whether measures were computed (using std_precomp) only after it tries loading the same measures to look for interpolated channels in lines 246-253.

CPernet, 2021-07-27:
std_limo()  should PR to https://github.com/sccn/eeglab/tree/develop
@arnodelorme that typo should be in next week release though ..

NirOfir, 2021-07-27:
Ok, I hope I made the PR correctly (I'm used to the way it works in FieldTrip's github, not sure if it's the right approach here). I now understand that the other issue I mentioned was already fixed in February...

Thanks!

CPernet, 2021-07-27:
+1 all good

PR #359 (2021-07-27): Fix typo in std_limo() default settings
Fixes #358.
closed 2021-07-27 by CPernet