sccn/eeglab #362: Loading issue from Paul

arnodelorme, 2021-08-06:
I’m trying to re-precompute channels measures for ERP data for a new STUDY secondary to adding new subjects and all of a sudden things are freezing up after clicking ‘ok.’ Previously I would see all the output processes on the Matlab command window, so I can only assume nothing is completing since nothing shows up on screen. I have to ctrl+c many times to interrupt whatever loop is running. I tried letting it run about 20 min (way longer than ever needed before for only 32 subjects) but nothing changed.

P-Beach, 2021-08-06:
Computer: Macbook Air, 2017 13in with 1.8GHz intel core i5, 8 GB 1600 MHz DDR3 memory, macOS 10.14.6
Matlab 2019a, eeglab 2021.0

Basically, after MUCH troubleshooting, I've found that eeglab will not load or compute anything when dealing with a subject file that was included in a STUDY file I created. It will load .set files for that subject, just not the one that was included in the STUDY.

Under Bug362 I have included sample data from one subject - the file ending with "_prnd" is the file included in the STUDY. The file ending with "_ICA" is the subject's file just prior to pruning components (not included in the study). 

Obviously one solution would be to just re-create all my pruned datasets, but with 33 subjects, all with different components kept or not kept, this would be a huge hassle. 

Thanks for any help you can provide!

arnodelorme, 2021-08-06:
@cll008 would you mind to look into this?

The data is below

https://drive.google.com/drive/folders/1chENaMaF7ZuiqGnsGTZRRvXat3FOvEO9

cll008, 2021-08-09:
Looks like the prnd dataset is causing a eeg_checkset & pop_loadset loop. Digging deeper at the mo.

cll008, 2021-08-09:
Stuck here because `EEG.data` is always `in set file`
https://github.com/sccn/eeglab/blob/eb18587231f8956086944beea540a789d7da7834/functions/adminfunc/eeg_getdatact.m#L149-L151

cll008, 2021-08-09:
@P-Beach I have a temporary workaround for you while we track down / patch this bug.
https://github.com/sccn/eeglab/blob/eb18587231f8956086944beea540a789d7da7834/functions/popfunc/pop_loadset.m#L139

After this line add `TMPVAR.data = [TMPVAR.setname '.fdt']` before `end`

If you are currently working with other datasets (particularly if they actually have data in .set instead of a separate .fdt), I suggest making a separate copy of `pop_loadset.m` for this adjustment and adding that script to the path only when working with this dataset.

cll008, 2021-08-09:
I have some clues but not 100% sure whey `pop_saveset.m` is bugging out. @P-Beach, what version of EEGLAB were you using when this used to work without any problems? Thanks.

P-Beach, 2021-08-10:
Hi Clement,

Thanks for your help looking into this. I've tried it with 3 versions of EEGLAB: 2019.1, 2021.0, and 2021.1. All more or less have the same problem.

The latter two are identical in the issue - screen appears frozen with out output after the command is entered. 2019.1 repeats the command-line pop_loadset command in a loop visible on screen.

cll008, 2021-08-10:
@P-Beach Did you try the workaround I mentioned? 

Thanks for the info. Did you save the pruned datasets on an earlier version of EEGLAB? When you mention trying with the 3 different versions, that's only for the loading part right?

P-Beach, 2021-08-11:
I just tried it (eeglab 2021.0) and it seems to work for the subject file I forwarded as well as for a different one. I am now able to *start* pre-computation of channel measures as well -- though eeglab is telling me .fdt files are not present for some subjects (they are).

I did initially create pruned files using two different versions of eeglab, but I later re-saved them using 2021.0 upon the recommendation of another user who said they had a similar problem at one time. Unfortunately the problem persisted.

cll008, 2021-08-11:
Did these initial pruned files cause the same issue in the other two versions of eeglab before you resaved in 2021.0?

For the subjects that are still not working, please add a breakpoint on the line you added and check if TMPVAR.data points to the correct .fdt file.

P-Beach, 2021-08-11:
Actually I may have misspoke. I don't think I got through renaming them all when this issue started happening.

I'm afraid I don't know what you mean about the break.

cll008, 2021-08-11:
No worries, just making sure the workaround is working so you don't have to recreate your files.
I'm closing this issue for now because I don't have the info to track down what caused the files to be saved incorrectly to begin with. Thanks for all your info and help @P-Beach

closed 2021-08-11 by cll008