sccn/eeglab #363: EEGLAB not listed on the octave package index

tpapastylianou, 2021-08-06:
Dear EEGLAB maintainers,

I noticed your package does not appear on the octave package index. Please consider adding it! You can find the octave package index and instructions how to add your package here: https://gnu-octave.github.io/packages/

Ideally, packages should be provided in the octave package format, such that could be installed via octave's pkg command; however, this no longer a strict requirement, and packages with custom installation instructions are also accepted on the index, as long as they provide clear installation instructions. :)

arnodelorme, 2021-08-10:
Thank you, we are going to add it.

cll008, 2021-08-11:
https://github.com/gnu-octave/packages/pull/8

closed 2021-08-11 by cll008