sccn/eeglab #365: EEGLAB error in function pop_biosig() at line 119

shutli1127, 2021-08-09:
### Description

[*EEGLAB error in function pop_biosig() at line 119.
data链接：https://pan.baidu.com/s/1PFZrQ7IMTq-4YnGFSQ_8mA 
提取码：lar0*]

<br>
#### Steps to Reproduce

1. [*load EEGLAB v2021.1*]
2. [*FIle - Import data - Using the BIOSIG interface*]
3. [*Error in function pop_biosig() at line 119*]

<br>
#### Expected behavior:
[*I want to load .RAW file using EEGLAB*]

<br>
#### Actual behavior:
[*I used the plugin BIOSIG to load .RAW file using EEGLAB, but it showed EEGLAB error in function pop_biosig() at line 119*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*WIndows 10 64bit*] |
| Matlab version | [*MATLAB 2020b*] |
| EEGLAB version | [*eeglab v2021.1*] |

cll008, 2021-08-09:
Hi @shutli1127 could you please upload a sample .raw file to a google drive link, along with a screenshot or description about the exact EEGLAB error that occurs. Thanks.

shutli1127, 2021-08-10:
> Hi @shutli1127 could you please upload a sample .raw file to a google drive link, along with a screenshot or description about the exact EEGLAB error that occurs. Thanks.

https://drive.google.com/file/d/1-CuPdOEehT6RbTp7j0_8ufb78xu4KlBm/view?usp=sharing. Thank you very much. This is the sample data link. And the error is "Unsuported data format ".

cll008, 2021-08-11:
What amp / system did you use to collect the data? I think the .raw that biosig imports has to be from EGI

arnodelorme, 2021-08-12:
.raw is also an EGI format (simple binary).

Arno

arnodelorme, 2021-08-16:
This file needs the binary simple plugin of EGI (called EGI legacy). I was able to import the data.

closed 2021-08-16 by arnodelorme