sccn/eeglab #367: fMRIb plugin doubled

jdien07, 2021-08-16:
Hi,
   when I install the fMRIb plugin via “Manage EEGLAB extensions", it now installs two copies:

one in a directory named "fMRIb2.00" and another in a directory named "fmrib2.00 (Case Conflict)"

This in turn causes error messages and other issues, such as:

EEGLAB: adding "fMRIb" to the path; subfolders (if any) might be missing from the path

Joe

cll008, 2021-08-16:
Thanks for reporting the issue. I tried replicating but things installed correctly on EEGLAB 2021.1 and Win10.You can manually remove `..\eeglab\plugins\fmrib2.00` and the error should disappear. 

What version of EEGLAB, MATLAB, and OS are you using? Were you installing fMRIb (no previous versions) or were you updating from another version? I'm closing the issue for now but please respond with more info if you would like to help.

closed 2021-08-16 by cll008

arnodelorme, 2021-08-16:
Hi Joe,

I am on a Mac and Clement is on a PC. It works for us. Maybe you did something different. Please reopen the issue if you find the problem.

Arno

jdien07, 2021-08-17:
Interesting!  I did more testing.  I did a clean install with v2021.1 on Matlab 2021a on the pop_os distro of Ubuntu 21.04 on a Dropbox folder and this happened again, so it replicates.  But I tried it on OS X 10.15.7 but otherwise the same and it did not happen.  So it seems to be OS dependent.  Not sure what else I can do testingwise but let me know.

Cheers!

Joe

> 
> 
> Hi Joe,
> 
> I am on a Mac and Clement is on a PC. It works for us. Maybe you did something different. Please reopen the issue if you find the problem.
> 
> Arno

--------------------------------------------------------------------------------

Joseph Dien, PhD
Senior Research Scientist
Department of Human Development and Quantitative Methodology
University of Maryland, College Park
Editor-In-Chief of Biological Psychology (along with Thomas Ritz)
E-mail: ***@***.***
Cell Phone: 202-297-8117
http://joedien.com

cll008, 2021-08-17:
Thanks Joe. I'll take it from here and check it out on my unix box, maybe the zip needs to be renamed correctly...

cll008, 2021-08-17:
Confirming that on unix the `tmp.zip` from urlwrite contains both lowercase and correct-case folders.

cll008, 2021-08-18:
Thanks again for the report and your help Joe. Turns out the zip contained a duplicate folder (which is invisible to case insensitive OSs). Reuploading the correct plug-in now, changes may take some time to reflect.

jdien07, 2021-08-18:
interesting!  didn’t realize OSes differed in case-sensitivity.  thanks for the update!

Joe

> 
> 
> Thanks again for the report and your help Joe. Turns out the zip contained a duplicate folder (which is invisible to case insensitive OSs). Reuploading the correct plug-in now, changes may take some time to reflect.

--------------------------------------------------------------------------------

Joseph Dien, PhD
Senior Research Scientist
Department of Human Development and Quantitative Methodology
University of Maryland, College Park
Editor-In-Chief of Biological Psychology (along with Thomas Ritz)
E-mail: ***@***.***
Cell Phone: 202-297-8117
http://joedien.com