sccn/eeglab #368: fMRIb plugin doubled (#367) reopened

jdien07, 2021-08-17:
Since you said to reopen it if I could replicate the problem, I did more testing.  I did a clean install with v2021.1 on Matlab 2021a on the pop_os distro of Ubuntu 21.04 on a Dropbox folder and this happened again, so it replicates.  But I tried it on OS X 10.15.7 but otherwise the same and it did not happen.  So it seems to be OS dependent.  

Under the hypothesis that Dropbox might have something to do with it, I tried it out again on the Linux computer, but this time in the Downloads directory.  Here, the directories were again doubled, but this time without the "(Case Conflict)" annotation.  So here the two versions were named:

fmrib2.00
fMRIb2.00

Let me know if there is anything further I can do to be helpful.

Cheers!

Joe

jdien07, 2021-08-17:
Further investigation online confirms that the "Case Conflict" annotation is provided by Dropbox since Linux is case-sensitive but Windows is not so it is intended to prevent problems when sharing files across platforms.  The reason why there are two different copies of fmrib with different cases when installing with the Extensions Manager on a clean (just unzipped) copy of EEGlab remains unclear though.

cll008, 2021-08-23:
Thanks for the additional testing Joe. Actually the issue is the fmrib2.00 zip itself, which contains a duplicate (I left some notes in the other closed bug report). This should be fixed now (I uploaded a new copy of the plug-in which only has 1 folder (instead of 2 folders with 'different' names).

closed 2021-08-23 by cll008