sccn/eeglab #369: file not properly imported from bdf files with EEGlab v2021.1

shabnamayat, 2021-08-23:
Hello,

I'm currently using the latest version of EEGLAB and Matlab R2018b. When importing a BDF file from Biosemi (BIOSIG toolbox),  I receive multiple warnings. A list of these errors can be found below:

Reading data file header...
Warning SOPEN: number of event onset (TYP=7C00) and event offset (TYP=FC00) differ (0,1)
WE HIGHLY RECOMMEND THAT YOU CHOOSE A REFERENCE CHANNEL IF THESE ARE BIOSEMI DATA
(E.G., A MASTOID OR OTHER CHANNEL). OTHERWISE, THE DATA WILL LOSE 40 DB OF SNR!
For more information, see http://www.biosemi.com/faq/cms&drl.htm
Warning SOPEN: number of event onset (TYP=7C00) and event offset (TYP=FC00) differ (0,1)
Warning SOPEN: number of event onset (TYP=7C01) and event offset (TYP=FC01) differ (0,89)
Warning SOPEN: number of event onset (TYP=7C02) and event offset (TYP=FC02) differ (0,96)
Warning SOPEN: number of event onset (TYP=7C03) and event offset (TYP=FC03) differ (0,113)
Warning SOPEN: number of event onset (TYP=7C04) and event offset (TYP=FC04) differ (0,109)
Warning SOPEN: number of event onset (TYP=7C08) and event offset (TYP=FC08) differ (0,30)
Warning SOPEN: number of event onset (TYP=7C16) and event offset (TYP=FC16) differ (0,156)
Warning SOPEN: number of event onset (TYP=7C17) and event offset (TYP=FC17) differ (0,156)
Warning SOPEN: number of event onset (TYP=7C1C) and event offset (TYP=FC1C) differ (0,39)
Warning SOPEN: number of event onset (TYP=7C34) and event offset (TYP=FC34) differ (0,63)
Warning SOPEN: number of event onset (TYP=7C35) and event offset (TYP=FC35) differ (0,47)
Warning SOPEN: number of event onset (TYP=7C3A) and event offset (TYP=FC3A) differ (0,10)
Warning SOPEN: number of event onset (TYP=7C3D) and event offset (TYP=FC3D) differ (0,71)
Warning SOPEN: number of event onset (TYP=7C3E) and event offset (TYP=FC3E) differ (0,1)
Warning SOPEN: number of event onset (TYP=7C40) and event offset (TYP=FC40) differ (0,51)
Warning SOPEN: number of event onset (TYP=7C65) and event offset (TYP=FC65) differ (0,89)
Warning SOPEN: number of event onset (TYP=7C66) and event offset (TYP=FC66) differ (0,96)
Warning SOPEN: number of event onset (TYP=7C67) and event offset (TYP=FC67) differ (0,113)
Warning SOPEN: number of event onset (TYP=7C7A) and event offset (TYP=FC7A) differ (0,156)
Warning SOPEN: number of event onset (TYP=7C7D) and event offset (TYP=FC7D) differ (0,156)
Warning SOPEN: number of event onset (TYP=7C98) and event offset (TYP=FC98) differ (0,63)
Warning SOPEN: number of event onset (TYP=7C99) and event offset (TYP=FC99) differ (0,47)
Warning SOPEN: number of event onset (TYP=7CA1) and event offset (TYP=FCA1) differ (0,71)
Warning SOPEN: number of event onset (TYP=7CA2) and event offset (TYP=FCA2) differ (0,1)
Warning SOPEN: number of event onset (TYP=7CFE) and event offset (TYP=FCFE) differ (0,29)
Warning SOPEN: number of event onset (TYP=7CFF) and event offset (TYP=FCFF) differ (0,25)
Reading data in BDF format...
eeg_checkset note: upper time limit (xmax) adjusted so (xmax-xmin)*srate+1 = number of frames
Importing data events...
eeg_checkset note: creating the original event table (EEG.urevent)
Re-referencing...
Re-referencing data

Here is a link to the file : https://www.dropbox.com/sh/xbhpj6atq1jo3pf/AAALA34Yj8tZpPpmIC2zyMCFa?dl=0

Please kindlly reply by considering me a beginner to the EEG analysis. Your assistance in this matter would be highly appreciated.

cll008, 2021-08-30:
Hi @shabnamayat, the error messages are saying that the number of event `onsets` do not match with the number of `offsets`. Do you know what is expected from the data? It's possible that this warning can be ignored, the way to find out is to check the `EEG.event` structure,

What kind of experiment is this? Looks like there is a consistent and strong artifact following one event.

arnodelorme, 2021-08-30:
@schloegl do you know about these warnings?

schloegl, 2021-08-31:
The warnings are there because:
1) Biosig represents event in an event table, with a rather strict representation of events, using position POS and TYP, or alternatively POS, TYP, DUR, CHN. (see also http://pub.ist.ac.at/~schloegl/matlab/eeg/EventCodes.html )
2) the decoding of these events from the BDF annotation channels is not always clearly defined, but some ambiguities can occur (e.g. if you have a bit field represenation or a state respresentation of the input,  e.g. [0,1,0,0,2,0,0,0,3,0,0,0,1,0,0,0] or [0,1,1,1,2,2,2,2,3,3,3,3,1,1,1] ).

In the present case, the default decoding mode is probably not want you want, but you might want to use mode 5 or 6, or even a different approach. You can check with the following commands whether any of these decoding approaches would match your expectations: 
```
     H=sopen('VP01.bdf'); H=sclose(H);
     more on 
     H5 = hdr2ascii(bdf2biosig_events(H,5);
     H6 = hdr2ascii(bdf2biosig_events(H,6);
     help bdf2biosig_events
```
If none of supported modes match your expectation, the decoding function might need to be extended for your needs. In that case, plase let me know which decoding result you expect.

closed 2021-09-07 by cll008

ashermanoxy, 2022-06-29:
Hi there,

I am having this same issue with my bdf data currently using EEGLAB v2022 in Matlab 2022. I'm struggling to understand how to apply a different biosig decoding mode into eeglab. When we record the data, we use NBS presentation to send the event codes. In Actiview, the correct event code labels are present. Thanks for your help.

Lucreziadimarino, 2024-10-21:
> Ciao,
> 
> Ho lo stesso problema con i miei dati bdf che attualmente utilizzano EEGLAB v2022 in Matlab 2022. Faccio fatica a capire come applicare una diversa modalità di decodifica biosig in eeglab. Quando registriamo i dati, utilizziamo la presentazione NBS per inviare i codici evento. In Actiview, sono presenti le etichette corrette del codice evento. Grazie per il tuo aiuto.

@ahermanoxy hi, i have the same problem with my .edf file... Have you resolved?

schloegl, 2024-12-30:
you can solve this by searching for the file "bdf2biosig_events.m", and change the default event decoding mode to your needs. 

```
if nargin<2 || isempty(Mode),
    Mode = 4; 
end; 

```

The alternative is that you ask eeglab to provide means to select the event decoding.