sccn/eeglab #379: ERSP plotting-Error in function newtimefbaseln() at line 112

jacobsen-noelle, 2021-09-06:
### Description

EEGLAB error in function newtimefbaseln() at line 112:
Index in position 2 exceeds array bounds.
[image]

It seems like the x input to this function is empty, which is why there's an error with the index exceeding array bounds. 

<br>
#### Steps to Reproduce

1. Load STUDY, datasets have ICA decomp
2.  STUDY--> precompute component measures --> spectra, ERSPs, default settings ('cycles', [3 0.8], 'nfreqs', 100, 'ntimesout', 200)
        -Test ERSPs works
3. Study --> Build preclustering array --> selected location based info : dipole locations, dipole orientation
4. Study -->Cluster Components, 15 clusters
5. Plot ERSPs using GUI

<br>
#### Expected behavior:
I would expect the ERSPs to plot just like they do in the Test ersp or with tftopo()

<br>
#### Actual behavior:
Error in newtimefbaseln() at line 112

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2018a] |
| EEGLAB version | [2021.0] |

arnodelorme, 2021-09-07:
Would you mind to upload your data (or send a separate link). If the whole study is too large, just upload a couple of subjects.

jacobsen-noelle, 2021-09-09:
I think the issue may have been that I had a field called EEG.event.condition, and it was conflicting with EEG.condition (which is empty). The function may be trying to find my ersp data at an empty index indicated by EEG.condition. I am going to see if changing the event field name to EEG.event.cond will help avoid this confusion. I was not aware this would affect ERSPs. I will let you know if this is the cause of the issue.

arnodelorme, 2021-09-20:
Yes, please let us know. Closing this issue for now but feel free to reopen.

closed 2021-09-20 by arnodelorme