sccn/eeglab #383: Error in std_precomp.m

Egofritte, 2021-09-20:
### Description

Hello,
i was trying to precompute channel measures (power spectrum) in a STUDY with two datasets of continuous data from the same subject. But instead of channel measures, i got the error attached in the file. As it is kind  of simliar to the error reported here (https://github.com/sccn/eeglab/issues/147), i tried the fixed version of the function std_precomp.m (https://github.com/sccn/eeglab/blob/develop/functions/studyfunc/std_precomp.m) but the error remained. Also looked up 'Preferences' if parallel toolbox was checked, which wasn't.

<br>
#### Steps to Reproduce

1. Loaded both datasets in STUDY and created the STUDY.
2. Clicked precompute channel measures, checked 'Power spectrum', and clicked 'OK'.
3. Error appeared.

<br>
#### Expected behavior:
Computation of channel measures.

<br>
#### Actual behavior:
Function std_precomp.m stopped and turned back an error.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 7 64-bit] |
| Matlab version | [2020b] |
| EEGLAB version | [2021.1] |
[image]

cll008, 2021-09-20:
Hi @Egofritte can you type 'edit std_precomp' and show me what line 234 looks like? When you use the function from issue #147 does the new error show at a different line?

Egofritte, 2021-09-21:
Hi Clement,

i added the additional code from @arnodelorme in https://github.com/sccn/eeglab/commit/30f2b88f7462b5f126215aa702bd404c72f309e9 to 'std_precomp'. The error remained the same. But after installing the 'Parallel Toolbox' the error resolved. However, i do not have checked the respective option in 'Preferences'.
So i need this toolbox to run the function, even though i dont use this toolbox?

cll008, 2021-09-21:
Interesting... It doesn't seem to make sense for the error to be the same (at least the line number should be different)? Could you  also set a breakpoint at 233? It seems like `option_parallel` should be 0 if you don't have the toolbox (or if you have the option set to not use toolboxes) and that `gcp()` should not be called.

Ideally the fxn works regardless of whether the toolbox is installed or not (since some people don't have access).

closed 2021-09-27 by cll008

arnodelorme, 2021-09-27:
This is because you do not have the parallel processing toolbox. Would you mind to overwrite this function and let us know if it fixes the problem.

https://github.com/sccn/eeglab/blob/develop/functions/studyfunc/std_precomp.m

Egofritte, 2021-09-28:
Hi @arnodelorme, 
i deinstalled the parallel processing toolbox to check whether the function now works with the changes you made. The result is a new error at line 468:
[image]

I thought the fxn should work regardless of whether the toolbox is installed or not (see @cll008 comment above)?! I am lucky that my institution provides a lot of toolboxes that i can use, but some people dont have this access.

helentobback, 2024-01-03:
I am stuck with the same problem and didn't find any solution yet in this. Is this issue still present in eeglab?

arnodelorme, 2024-01-03:
Which version of EEGLAB are you using?

helentobback, 2024-01-04:
Dear Mr. Delorme,

I am currently EEGLAB 2023.0. Is it necessary to update the version? I’ve done this earlier, but my scripts wouldn’t even allow the import of several datasets (too many input arguments error), so I put back the old version to continue with my current scripts.

Vriendelijke groeten

***@***.***<http://www.vub.ac.be/>

Helen Tobback
Doctoral researcher and Teaching assistant
Faculty of Psychology and Educational Sciences
E   ***@***.******@***.***>
T   +32 (0)2 629 22 91
Room B3.20 - Pleinlaan 2 - 1050 Brussel
www.vub.be<http://www.vub.be/>

HET DENKEN ONDERWERPT ZICH NIET   #WeAreVUB

Schrijf je in voor onze nieuwsbrief <http://communicate.vub.ac.be/subscription-center-nl> -  Sign up for our newsletter<http://communicate.vub.ac.be/subscription-center-en> - disclaimer<http://huisstijl.vub.ac.be/styleguide/disclaimer>

helentobback, 2024-01-04:
FYI in the command window the first of the failed actions of the first step of LIMO is the job design and it gives the next error: Unrecognized field name cellArray.

Vriendelijke groeten

***@***.***<http://www.vub.ac.be/>

Helen Tobback
Doctoral researcher and Teaching assistant
Faculty of Psychology and Educational Sciences
E   ***@***.******@***.***>
T   +32 (0)2 629 22 91
Room B3.20 - Pleinlaan 2 - 1050 Brussel
www.vub.be<http://www.vub.be/>

HET DENKEN ONDERWERPT ZICH NIET   #WeAreVUB

Schrijf je in voor onze nieuwsbrief <http://communicate.vub.ac.be/subscription-center-nl> -  Sign up for our newsletter<http://communicate.vub.ac.be/subscription-center-en> - disclaimer<http://huisstijl.vub.ac.be/styleguide/disclaimer>