sccn/eeglab #384: Error to import Neuroscan .CNT file

VictorGimenez, 2021-09-21:
### Description

[*I am facing an issue at the moment when I try to upload a .CNT extension file through: File > Import Data > Using EEGLAB functions and plugins >  From Neuroscan .CNT file, it is showing me that there is an error at the loadcnt() at line 'str2mat' undefined near line 693 column .*]

https://drive.google.com/file/d/1mLi-WPdhjwrSzdhWcUO_qJPYj9zrMcik/view?usp=sharing

<br>
#### Steps to Reproduce

1. [*Open the folder where EEGLAB/eeglabdevelop was uncompressed*]
2. [*Open Octave through line command on terminal typing: octave --gui*]
3. [*Typed eeglab at the command line*]
4. [*At EEGLab select File tab*]
5. [*At File tab select Import data*]
6. [*At Import data tab select Using EEGLAB functions and plugins*]
7. [*At Using EEGLAB functions and plugins tab select From Neuroscan .CNT file*]
8. [*Select the .CNT file*]
9. [*Data format set to Autodetect*]
10. [*Checked the checkbox Check to import keystrokes*]
11. [*OK*]

<br>
#### Expected behavior:
[*Next screen*]

<br>
#### Actual behavior:
[*Message EEGLAB error in function loadgnt() at line 'str2mat' undefined near line 693 column...*]

<br>
#### Versions

|   |   |
|----| ---- | 
| Ubuntu version     | [*20.04.2 LTS Focal Fossa*] |
| Octave version | [*5.2.0*] |
| EEGLAB version | [*v2021.1*] |

VictorGimenez, 2021-09-23:
How could I proceed? Thanks

VictorGimenez, 2021-09-27:
I am very stuck on it, and there isn't another version about the neuroscanio package to download, it looks like be a issue involving anything on this function `loadcnt()`, the picture was private before I realized it but now it's public to check it out.

cll008, 2021-09-27:
Hi Victor, thanks for your patience.
If we check 'help str2mat' in the Octave command line we find that it is a file that is not yet implemented.
Please try the following:

1. Edit loadcnt to open up the script
2. Navigate to line 693 where `str2mat` is
3. Replace `str2mat` with `char`
4. Save the changes to the script
5. Try to load file again

Let me know if that works.

VictorGimenez, 2021-09-27:
Hey @cll008 thank you so much for the reply!! It worked until now and I'm in the Dataset info -- pop_newset() window, I hope that after this fix I am not gonna receive more errors!!

cll008, 2021-09-27:
Awesome. I'll close this for now but feel free to follow up here or open another issue should one arise.

closed 2021-09-27 by cll008